robotology/community #1: Name for the Q&A virtual repo

pattacini, 2015-01-12:
Q&A Kick-off.
Any ideas about the name?

cc @lornat75 @paulfitz @randaz81 @vtikha

lornat75, 2015-01-12:
I am in favor of this, but I think we should keep the discussion the mailing list for another bit.

lornat75, 2015-01-12:
@drdanz do we need a specific team/group to allow people to post issues?

alecive, 2015-01-12:
@lornat75 as far as I know, any github user can post an issue in any github public repository, so I would say no :)

kt10aan, 2015-01-12:
I think "Support" is more self explanatory :)

Any github user can post an issue in any github public repository, however, for a user to set a label for that issue, he has to be in the "contributors" group.

pattacini, 2015-01-12:
creating new labels should be put under strict control, indeed :)

lornat75, 2015-01-12:
@alecive, @kt10aan noted in: d639cd58596b3b9f4c24056492b95954ee85c063

alecive, 2015-01-12:
@pattacini problem is that only the `owners` group can assign labels to issues. e.g. I cannot assign any label to this issue even though I am both in the `YARP developers` and in the `iCub developers` group.

drdanz, 2015-01-12:
@lornat75 confirming @alecive reply, any GitHub user can post issues on a public repository

pattacini, 2015-01-12:
@alecive I think restraining to `owners` is ok indeed :)

alecive, 2015-01-12:
@pattacini I think the same, but that means that you guys have to manually label any new issue because anybody else cannot do that. It can become quite annoying after a while.
My point was that labeling is a powerful process, and it would be nice if issues were already labeled by the users, instead of being labeled by the contributors. But I don't think that we have another choice.

pattacini, 2015-01-12:
new labels can be proposed and then, once they get accepted, attached to open issues

pattacini, 2015-01-12:
It seems this post has some information about the _labeling_ issue:
https://groups.google.com/forum/#!topic/tardis-devel/6IQCFJxxROk

alecive, 2015-01-12:
The quickest way would be to create another group, e.g. `robotology-collaborators`, gradually put everyone in it, and assign full admin access to this repository (i.e. only `robotology/QA`). A bit risky, though it is a manageable risk.

pattacini, 2015-01-12:
I've created the team `qa-users` with _write-access_ to address the labeling issue; we need to populate it with users...

drdanz, 2015-01-16:
In case we want to import issues from other repositories to the QA one, this is a nice tool: https://github.com/IQAndreas/github-issues-import

closed 2020-12-08 by pattacini