robotology/community #2: MC4/MCP boards Openloop measurements

Farinha91, 2015-01-14:
Hi all,

I'm currently working with a MC4-001 control board to control 4 DC motors w/encoders using yarp and the icub repository and I would like to optimize the PID values stated in the .xml configuration file for my robot.

In order to do that I would like to first characterize the system open loop behaviour. For that purpose I'm already trying out the robotMotorGui with the open loop option and it works, but I'm having 2 problems regarding that approach:

- I would like to be able to read the actual and the desired open-loop "values" or voltages in order to make a plot of both, but it is not possible through the GUI. Would this be possible with an oscilloscope? If it is, what pins should I read? If not, would it be better to make this through code?

- In the robotMotorGui, with a joint set to openloop, when I set the "Desired openloop value" (joint PID button->Openloop params tab) to 80 it moves about 200 degrees, as read in the GUI, but if I set the Desired openloop value to 90 it moves (apparently) to infinity (the encoders are already counting 10000 degrees). I'm not really familiar with control but isn't this strange?

Thanks for the attention given,
Best regards,
André Farinha.

pattacini, 2015-01-14:
You might find useful some work we carried out in the past regarding the online tuning of PID controllers. It was done specifically to improve the control performance of fingers, but it can be extended without limitation to a generic DC motor.

Resources:

The report we wrote up is available here: [pid-tuning.pdf | uploaded via ZenHub](https://files.zenhub.io/54b65e1aa249c8781446c7a6).

The code referred in the document is in the [`ctrlLib`](http://wiki.icub.org/iCub/main/dox/html/group__Tuning.html) library belonging to the `icub-main` bundle, which is then linked by the [`fingersTuner`](http://wiki.icub.org/iCub/main/dox/html/group__icub__fingersTuner.html) tool.

Hope this helps.

Farinha91, 2015-01-14:
Hi, thanks for the answer, however, as this is for my Master Thesis (I'm building a robotic hand) I would first like to try to do this by hand as I feel I can learn a bit more from it. 

In that sense I already did a simple script where, using yarp::dev::IControlMode2 and yarp::dev::ImplementOpenLoopControl, I can send and read commands from the motors in openloop mode, using setRefOutputs and getOutputs.

Up to this point everything goes smoothly but, contrary to the stated in the getOutputs function description, that function immediately returns the value of the final "position" with the motor still moving to get to that point. Basically it is behaving like the getRefOutputs function is expected to, returning the last reference sent using the setRefOutputs function. I just needed something that showed, at each time, the value of the controller and how far it is from the requested "position", similar to how the encoders work in a position control scheme. Is that possible?

Thanks,
André

pattacini, 2015-01-14:
You should simply impose a voltage reference to the motor in open-loop mode - as you're actually doing - by calling `IOpenLoopControl::setRefOutput()` and then you can read back the current joint position by calling the method `IEncoders::getEncoder()`. Both `IOpenLoopControl` and `IEncoders` are interfaces than need to be opened as _views_ over the same device.

```cpp
PolyDriver drv;
IOpenLoopControl *iolc;
IEncoders *ienc;

// here you open the remote_controlboard device drv

drv.view(iolc);
drv.view(ienc);
```

Farinha91, 2015-01-14:
Ok, but then is there anyway to convert the voltage reference I'm imposing to the angle I should be expecting?

pattacini, 2015-01-14:
To a certain extent the DC motor can be modeled with the following transfer function: y(s)/u(s)=K/(s(1+s*tau)). Applying a voltage will elicit a speed, not a constant position.

Remember also that the voltage is expressed in dimensionaless manner by specifying ticks over a given scale. The current available scales, as far as I know, are: [-800, +800] or [-1333, +1333] ticks. This should be defined somewhere in the configuration file.

pattacini, 2015-01-20:
@Farinha91 please consider closing the issue or elaborating more on it in case you need further help.

Farinha91, 2015-01-20:
Sorry for the delay, I was expecting that applying a voltage to the motor would set a constant position and not a speed, so my original idea of applying the root locus to optimize the systems position PID is not applicable.

I will try to apply the Zigler-Nichols method to optimized the PID values for position control. Thanks for the help anyway.

pattacini, 2015-01-20:
Keep in mind that identifying the system response to some peculiar input commands (e.g. step-wise references, chirps, sinusoidal signals at varying frequencies) is always a better approach than relying on heuristic methodologies.

Farinha91, 2015-01-21:
Just to check if I understood, what you proposed is that I try different values for the Kp, Ki and Kd gains of the PID and check which set of values guarantees a better response to those input commands, right?

I'm just asking because I've already been proposed to study/characterize the response of the system (its static gain and pole) without feedback in order to apply a root locus afterwards and after what we have talked I'm afraid that it is not possible however I'm don't know enough about control to be sure...

Sorry for bothering,
André

pattacini, 2015-01-21:
Nope, what I meant is that during identification you deal with open-loop response of the plant, whereby you command a voltage and you measure back a joint position, thus the PID is not considered at all. For instance, you can command a square wave of as input voltage and so you'll get a triangular wave as output position. The plant characterization is still possible then.

In the [document](https://files.zenhub.io/54b65e1aa249c8781446c7a6) I posted [above](https://github.com/robotology/QA/issues/2#issuecomment-69908682) you can find a more detailed answer.

Farinha91, 2015-01-21:
Ok, I'll then read the document and close this for now. Thanks for the attention.

Farinha91, 2015-01-22:
Hi. I read the document posted above and it was really helpful, sorry for overlooking that in the first time.

However I'm now trying to use the fingersTuner and I'm having a problem. I have a robot description that is similar to the one in [here](https://github.com/robotology/icub-main/tree/master/app/robots/experimentalSetups/single_joint_MC4) (only changed the PID parameters, while using it as icub/left_arm) and I am able to move the motors with robotInterface + robotMotorGui. Up to here everything is ok, however, when I run the robotInterface + fingersTuner I get, from fingersTuner:

yarp: Port /fingersTuner/1/left_arm/rpc:o active at tcp://10.0.28.83:49487
yarp: Port /fingersTuner/1/left_arm/command:o active at tcp://10.0.28.83:60306
yarp: Port /fingersTuner/1/left_arm/state:i active at tcp://10.0.28.83:42997
yarp: Port /fingersTuner/1/left_arm/stateExt:i active at tcp://10.0.28.83:40179
yarp: Sending output from /fingersTuner/1/left_arm/rpc:o to /icub/left_arm/rpc:i using tcp
yarp: Sending output from /fingersTuner/1/left_arm/command:o to /icub/left_arm/command:i using tcp
yarp: Receiving input from /icub/left_arm/state:o to /fingersTuner/1/left_arm/state:i using udp
yarp: Receiving input from /icub/left_arm/stateExt:o to /fingersTuner/1/left_arm/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
[INFO]Checking if left_arm is active ... yes
yarp: Port /fingersTuner/2/right_arm/rpc:o active at tcp://10.0.28.83:38799
yarp: Port /fingersTuner/2/right_arm/command:o active at tcp://10.0.28.83:51893
yarp: Port /fingersTuner/2/right_arm/state:i active at tcp://10.0.28.83:57489
yarp: Port /fingersTuner/2/right_arm/stateExt:i active at tcp://10.0.28.83:41504
Problem connecting to /icub/right_arm/rpc:i, is the remote device available?
Problem connecting to /icub/right_arm/command:i, is the remote device available?
Problem connecting to /icub/right_arm/stateExt:o from /fingersTuner/2/right_arm/state:i, is the remote device available?
yarpdev: ***ERROR*** driver <remote_controlboard> was found but could not open
[INFO]Checking if right_arm is active ... not yet: still 39.9 [s] to timeout expiry

and robotInterface (most of the times) exits with a segmentation fault.

I currently have the firmware version 1.11 build 80.

While it seems the first connection to the left_arm is successful I don't know where he is getting the right_arm reference... Or why does the robotInterface gives segmentation fault... Any hint?

pattacini, 2015-01-22:
How is your setup made of? Do you have a complete robot or just a motor?
In case you have only one arm, you can disable the missing arm from the `fingersTuner` configuration file by correctly populating the list `relevantParts` under the `[general]` section.

Farinha91, 2015-01-22:
I have a pair MC4/MCP controlling 4 motors, just that.

Already found the right config file and was already able to try to tune 2 joints. It is so cool :o!
I just noted that in both tries the stiction results came as 0, however even before I had already noted some stiction "problems" with the motors. I already tried to check, but is there any parameter to activate this analysis or something that is stopping it from having results?

pattacini, 2015-01-22:
The `encs_ratio` serve to convert the PID gains as identified by the tuner into proper values that in turn can be used by the firmware. Basically, we have to account for [degrees-to-encoders_ticks conversion](https://github.com/robotology/icub-main/blob/master/src/tools/fingersTuner/main.cpp#L279) and these ratios cannot be retrieved at run time via yarp; that's why we need to specify them within the configuration file.

Concerning the stiction estimation, does the joint actually move during such a phase? It is commanded in open loop.

Farinha91, 2015-01-22:
Oh, so the encs_ratio is the same as the encoders parameter in the robot hardware xml. Okay, thank you.

Regarding the stiction estimation, yes, the joints moves during the 60 seconds but in the end the stiction values both come as 0...

pattacini, 2015-01-22:
Then it might be that the default parameters used for tuning fingers are not suitable to run with other motors.

The state machine responsible for managing such identification is sketched out in Figure 11 of technical report and described thereafter, whereas it is implemented by this [run() method](https://github.com/robotology/icub-main/blob/master/src/libraries/ctrlLib/src/tuning.cpp#L321) in the `ctrlLib` library.

I would first increase `vel_thres` as well as decrease `e_thres` parameters specified at this [line](https://github.com/robotology/icub-main/blob/master/src/tools/fingersTuner/main.cpp#L223) to see whether these changes will affect the outcome. As second attempt, I would start inserting some debug printouts in the `run()` method referred above in order to figure out what is going on.

pattacini, 2015-01-22:
Consider also that this specific stiction identification procedure turns to be effective when stiction component is very dominant at low velocity (i.e. fingers with tendons-sheath actuation), while in most cases friction parameters can be found relying on least-squares approach (both batch and recursive).

Farinha91, 2015-01-22:
Well, I tried both suggestions and while it was confirmed the run() method was running fine the stiction identification still returned 0.

Despite all this I checked that the motors were moving with no apparent difficulty during the stiction measurement. Furthermore, my previous observations on stiction regarded the openloop control of the system where I needed to send a command higher than 70-80 to get the motors moving (overcome the friction), which comparing to a max command of 1300 is not very significant. 

Also, I don't yet have the motor connected to the finger (through the tendon) and after that connection is made the effect of the stiction should be more relevant. I'll test again after that but as long as I see the motors moving through the stiction measurement without any apparent problem (or without additional problems) I won't give it much importance.

I'll just leave the topic open until I can test the motor connected to the finger (should happen soon).

Thanks,
André

pattacini, 2015-01-22:
Well, 70-80 PWM ticks are not that negligible compared to the full scale; however, the stiction identification observes the position errors computed while tracking a quasi minimum-jerk profile under the action of a P-controller commanding the voltage. This position error is integrated when low velocity condition is detected. This is to say that, for some reason, it is likely that such adaption might not get triggered at all. To that purpose I have mentioned the printouts to be put in the `run()` method, that is just to check whether the `adapt` variable is set to _true_ at least once.

Anyway you're right, it's better to wait for the transmission with the joint to be made.

Farinha91, 2015-01-30:
Sorry for taking so long to close this topic, our printer is currently broken...

Meanwhile, in the report that was referred it said that the differential component of the PID was implemented using a FIR filter and therefore the pole placement of the closed loop was not possible by the user. I tried to find why but still don't know, would you mind explaining? Also, just to be completely sure, this precludes the application of methods as root loccus or not really? 

Thanks for the attention,
Best regards,
André Farinha

pattacini, 2015-01-31:
The hitch is that a discrete FIR filter has no simple continuous counterpart and thus cannot be directly remapped into standard tools used for PID design.

A pretty nice tutorial illustrating those tools can be browsed [here](http://ctms.engin.umich.edu/CTMS/index.php?example=MotorPosition&section=ControlRootLocus). It covers also the _root locus_ design. The PI controller can be still approached with the root locus, I'd say.

pattacini, 2015-02-10:
A somehow related issue: https://github.com/tacman-fp7/tactile-control/issues/3

closed 2015-02-11 by pattacini

pattacini, 2017-05-30:
@S-Dafarra this is the issue we talked about this morning on the tuning of a PI controller.