robotology/community #3: Can I cross-reference an issue from a different repository (i.e. in robotology)?

lornat75, 2015-01-15:


lornat75, 2015-01-15:
Yes, you can include the name of the repository before the name, i.e.

This is dealt with in robotology/<repository-name>#<issue_number>

closed 2015-01-15 by lornat75

traversaro, 2015-01-15:
Trick that saves some time, copy pasting directly the url of the issue result rendered in the same way, for example `https://github.com/robotology/codyco-superbuild/issues/43` becomes https://github.com/robotology/codyco-superbuild/issues/43 .

rbonghi, 2015-01-15:
This is a good guide to understand principal feature about MarkDown
https://guides.github.com/features/mastering-markdown/

About **GitHub Flavored Markdown**
https://guides.github.com/features/mastering-markdown/#GitHub-flavored-markdown