robotology/community #4: Hand moving spontaneously

robertocalandra, 2015-01-16:
During a pause between two experiments (using the WBI toolbox), two fingers of the left hand of iCubDarmstadt01 moved spontaneously (from completely extended to completely flexed).
I am not sure if this ticket is depending from the WBI-toolbox or not (might be lower-level).
This is quite awkward (and scary)...
What should I try to log next time to diagnose the possible problem?

DanielePucci, 2015-01-16:
The WBI and, as a consequence, the toolbox have nothing to do with the hands since  the version you are using is not configured to allow the control of the robot's hands, unless you specifically changed the source code. Sometimes it happens that the fingers can move a little without touching them, thus scaring a little bit the user :)

robertocalandra, 2015-01-16:
Hi Daniele,
Yes, I know that the WBI does not have direct access to control the hand. 
But the one I experienced was not a little movement (i.e., noise). It was a full flexion of two fingers. 
Since there are so many layers (Yarp, firmware,hardware), how can I find out what was the source of this spontaneous movement?

randaz81, 2015-01-20:
Some 'general' questions before starting any debug activity:
- Does the issue happen frequently or it happened just once? 
- Is WBI somehow involved (maybe just executed once and then stopped) in this kind of issue?
- Does it happen even just starting the robot from a clean configuration, without executing any user module?
- What is the status of the joint shown in the robotMotorGui? is the encoder value correct? is the joint still in position mode or it is an different mode (idle / hardware fault / under calibration etc).
- does the robotInterface log show any strange error/warning?
- does the log of other modules show any strange error/warning? With the yarplogger tool you can save the log of a entire session (meaning that you can retrieve the log of all the the processes you launched from the startup).
- is your system updated? which version of yarp/icub/fw/WBI toolbox are you using?

robertocalandra, 2015-01-22:
Ok, now I am pretty sure that it is a consequence of the WBI.

> Does the issue happen frequently or it happened just once? 
- The issue seems to happen regularly, and it is valid for both hands

> Is WBI somehow involved (maybe just executed once and then stopped) in this kind of issue?
- The WBI is definitely involved

> Does it happen even just starting the robot from a clean configuration, without executing any user module?
- it happens after using WBI, even if no additional module is required.

> What is the status of the joint shown in the robotMotorGui? is the encoder value correct? is the joint still in position mode or it is an different mode (idle / hardware fault / under calibration etc).
- The motore are still in position mode and properly working (the encoder value is correct)

>does the robotInterface log show any strange error/warning?
>does the log of other modules show any strange error/warning? With the yarplogger tool you can save the log of a entire session (meaning that you can retrieve the log of all the the processes you launched from the startup).
- I still need to run it

> is your system updated? which version of yarp/icub/fw/WBI toolbox are you using?
- I am using the previous version of the WBI. How can i check the versions for the other components? 

I am also attaching a video https://www.dropbox.com/s/sj0x0j3r96lxa64/VID_20150122_165355.mp4?dl=0

DanielePucci, 2015-01-22:
Hi Roberto,
nice video. Just to rule out some other implications, have you tried to write a simple C++ module that does the same operations of the WBI? This might be of some help.

traversaro, 2015-01-23:
That is definitely interesting. You where using the old (v0.1) or new (was new_wbi_ID branch, now v0.2) version of WBI-Toolbox? In the latter case, which configuration file you where using?

closed 2020-12-15 by pattacini