robotology/community #7: Right wrist is stuck

kt10aan, 2015-01-20:
Hi,

The right wrist of our iCub  (Hertfordshire01) is stuck. I mean that it doesn't move at all, even with no power, when the robot is off (the movements corresponding to joints 5 and 6 in the right arm).

The last thing I was doing was running some pretty stock stuff in the ARE. Then, when I turned it off, I noticed that in the parking position the wrist was still stiff.

RobotmotorGui reports the joints as operational (however, moving the slider for joints 5 and 6 doesn't make any movement.

I have rebooted the robot completely a few times.

The log is here (taken from the YarpLogger).
https://dl.dropboxusercontent.com/u/22756225/log_pc104_robotInterface_3361

Some errors that are showing are during the shutdown procedure, and not during the startup, and they are there forever :) (AXIS_OVERLOAD)

pattacini, 2015-01-20:
@kt10aan, for the time being it would be still better to forward such issues to mechies - since it seems the problem regards the mechanics - thus relying on icub-support@icub.iit.it. However we are about to harmonize Q&A also for them.

randaz81, 2015-01-20:
If a joint does not move at all, even with no power, most likely the corresponding motor is burned or the gearbox is stuck/broken.

kt10aan, 2015-01-20:
@randaz81, it doesn't sound good :( Is there a way to determine if it is the motor or the gearbox? How can this be fixed?

closed 2020-12-15 by pattacini