robotology/community #9: Getting joint limits

matusko89, 2015-02-01:
So after my  original psot on robotcub 
"Hi hackers, 
I was trying to find if there is a way to get the limit for some iCub simulator joint. Something that would return the lower or upper limit of desired joint. Is there a way to do this or I need to remember this limits for each joint?"
and reply from pattacini I am opening this issue :)

barbalberto, 2015-02-02:
To get the limits you have to get the IControlLimits interface (documented here http://wiki.icub.org/yarpdoc/classyarp_1_1dev_1_1IControlLimits.html) using the view command.

yarp::dev::IControlLimts iLim;
device.view(iLim);     // device is your remoteControlBoard

and then call the getLimits method
  iLim->getLimits(int j, double *min, double *max);

This will return the min & max for the given joint j.

pattacini, 2015-02-02:
To open a motor device to then use the `yarp::dev::IControlLimits` interface, refere to the [wiki](http://wiki.icub.org/iCub/main/dox/html/icub_motor_control_tutorial.html).

closed 2020-12-08 by pattacini