robotology/community #10: Support for CUDA parallel programming

gsaponaro, 2015-02-02:
Out of curiosity, which iCub modules take advantage of parallel computing frameworks (e.g. CUDA, OpenCL, OpenMP, TBB)?

It would be helpful to have a brief list, in order to know which modules (from the [icub-main applications list](https://github.com/robotology/icub-main/tree/master/app)) to deploy on the Nvidia machine of our cluster and which ones on the non-Nvidia machines. For example, I know that the ```motionCUT``` module takes advantage of CUDA.

pattacini, 2015-02-02:
`motionCUT` was used to take advantage of OpenMP while now it benefits from TBB, not CUDA at all.

[`icub-main`](https://github.com/robotology/icub-main) does not contain CUDA dependent code, if I'm not mistaken.
By contrast, repositories such as [`iol`](https://github.com/robotology/iol), [`stereo-vision`](https://github.com/robotology/stereo-vision) and [`himrep`](https://github.com/robotology/himrep) require the `SiftGPU` package and thus CUDA.

gsaponaro, 2015-02-02:
Thanks for the clarification.

Regarding ```motionCUT```, part of the reason I got confused among the different frameworks, was that it sets an internal macro with OpenMP in its name:
```
    // check if OpenCV supports OpenMP multi-threading
    #if (CV_MAJOR_VERSION>0) && (CV_MAJOR_VERSION<3)
        #define _MOTIONCUT_MULTITHREADING_OPENMP
    #endif
```
but I guess ```motionCUT``` is agnostic to whatever optimization OpenCV is configured/compiled to use.

pattacini, 2015-02-02:
Well, `motionCUT` is not completely agnostic, in a sense that it knows that some old versions of OpenCV were designed to work with OpenMP, while the most recent versions deal with the new TBB layer.

closed 2020-12-08 by pattacini