robotology/community #12: ActionsrRenderingEngine home position and gazing

kt10aan, 2015-02-09:
After the last updates in ARE (well, I do not know exactly in what point), there have been two changes in the behaviour:

1) The home position appears to be changed. While with regards to limbs and torso that is fine, the gaze/head home position appears a bit strange (i.e., where in the past it was looking straight more or less, now it achieves the same by lowering the head and the eyes going quite upwards. Is this intended? Is this on our robot only?

2) The gazing itself appears to be off. E.g. where in the past a given point was resulting to the expected behaviour, now the same given point results in the gaze being off, mostly in the z axis.

pattacini, 2015-02-10:
`ARE` is very unlikely the cause of such weird effect.
Are you using "saccades on" option for the `iKinGazeCtrl` module? I don't know what version of `icub-main` you are relying on, but lately a fix has been applied in order to make the fast saccades work again: see https://github.com/robotology/yarp/issues/387.

kt10aan, 2015-02-10:
As i mentioned in issue#11, I am using an icub-main version from December 2014. I will update and report any improvements :)

pattacini, 2015-02-18:
Closing the issue.
Feel free to reopen it if you have new info on that.

kt10aan, 2015-03-10:
Sorry for taking so long. I reverted back to iCub v1.1.5 (along with corresponding YARP and firmware-shared) and ARE (specifically, the gaze in ARE) is working as it should.
The date of the git version of the software with the wrong behaviour was 24th December. However, despite the wrong behaviour in ARE, the gazeController was working fine with a simple gaze module (e.g. with the gaze controller tutorial).
I went through the commits in ARE between July and December but I didn't see anything that could cause that behaviour. Of course, I might have missed something, or the issue was somewhere else.

pattacini, 2015-03-10:
Since the `head` in the repo is working I think that next upcoming package release of `icub-main` should fix the problem.

closed 2020-12-15 by pattacini