robotology/community #13: icub - cameras unfocused + vertical offset

Arikae, 2015-02-10:
We experienced some problems with the eyes of the icub.
Somehow the right eye is always unfocused and seems to have a vertical offset compared to the left eye (it looks more up than the left one).
Is this normal or can it be calibrated somehow?

pattacini, 2015-02-11:
@Arikae hold on, we are collecting precise information on how compensate for the vertical offset.

Regarding the camera focus; you could be able to manually rotate the camera lens to adjust the focus. This can be done by operating directly on the eyes if you have a version of the iCub mechanics < 2.0. Conversely, there is an inner screw preventing such a rotation and thus you would need to open the eyeball.

ale-git, 2015-02-11:
@Arikae about the vertical offset, please try to add  
--yoff <vert_offset> 
to your camera parameter list. Try different values until the two cameras are equal. Hope this helps.

robertocalandra, 2015-02-11:
@pattacini Do you know if there is a manual that shows the procedure to manually adjust  the focus? (icub v2.5)

pattacini, 2015-02-11:
For this, and for the time being, I'd suggest you to post on icub-support@icub.iit.it

Arikae, 2015-02-12:
@ale-git Alright, thanks for the hint. We'll try that out.

@pattacini Thanks for the reply, we'll try to contact them.

pattacini, 2015-02-18:
@Arikae @robertocalandra 

This [document](https://svn.robotology.eu/repos/iCubHardware-pub/trunk/mechanics/public/doc/howTo/iCub2_Camera_Alignment_Focus.pdf) recaps the procedures for aligning the camera and changing the focus.

robertocalandra, 2015-02-19:
Thank you @pattacini @ale-git :) 

 One more question since I am not an expert: `--yoff vert-offset` should be added to which module exactly? to `icubmoddev` or `yarpview` ?

pattacini, 2015-02-19:
The document mentions to add that parameter to the **camera configuration file**, not to specify it in the xml file. Anyway, `icubmoddev` is the one.

robertocalandra, 2015-02-19:
Ok, I see. But `--yoff vert-offset` is limited to 2 pixels (since the vertical resolution is 322 and 320 is the standard image size), correct?

I also saw that there is a camera configuration for a resolution 640x480. Is this camera already use on the icubs? To be more specific, would iCubDarmstadt01 be capable of higher resolutions?

Thank you

pattacini, 2015-02-19:
The `640x480` resolution can be used by any iCub. You will have different bandwidth load and different rate depending on the use of `bayer` carrier.

@ale-git  any thought on `--yoff vert-offset`?

ale-git, 2015-02-19:
In 320x240 the vertical resolution is 244, so you can set an offset between R an L camera up to four pixels. The offset is intended for small adjustments... if the difference between the images is too high, maybe there is some mechanical problem in the eyes?

robertocalandra, 2015-02-19:
Ok, thank you, it was just to confirm it. :)

We will then proceed with the mechanical calibration (in a couple of weeks).

closed 2020-12-15 by pattacini