robotology/community #15: Yarp autocomplete support

traversaro, 2015-02-20:
Hi everyone, 
  I know that the `yarp` and `yarp-config` command line tools has autocomplete support on *nix systems, but I don't know ho to enable it. Anyone knows some relevant documentation?

cc @rlober @darwinlau

drdanz, 2015-02-20:
Not sure if there is any documentation about... anyway: 

```bash
cp $YARP_ROOT/scripts/yarp_completion /etc/bash_completion.d/yarp
```

Feel free to add it somewhere in the documentation :)

traversaro, 2015-02-20:
Super cool, thanks! 
Added  this in Linux installation from sources tutorial, in http://wiki.icub.org/index.php?title=Linux:Installation_from_sources&diff=prev&oldid=20987 . (Perhaps should be added even in OS X one?).

drdanz, 2015-02-20:
@traversaro Actually
```sh
ln -s $YARP_ROOT/scripts/yarp_completion /etc/bash_completion.d/yarp
```
Is probably a better option, so that when the file is updated, you don't have to copy it again.

@francesco-romano does it work on OSX too?

I just checked and at the moment it does not work out of the box on cygwin, but it works if you run dos2unix on it, or if you add this before the completion script is executed (I added a file in ``/etc/profile.d`` because at the time of execution of the ``~/.bashrc`` file is too late).
```sh
(set -o igncr) 2>/dev/null && set -o igncr;
```

See also https://stackoverflow.com/questions/11616835/cygwin-r-command-not-found-bashrc-bash-profile

gsaponaro, 2015-02-20:
The page http://wiki.icub.org/wiki/Generic_iCub_machine_installation_instructions suggests to add the tab completion script to your .bashrc (or similar file):

    if [ -f $YARP_ROOT/scripts/yarp_completion ]; then
      source $YARP_ROOT/scripts/yarp_completion
    fi

kt10aan, 2015-02-20:
On OS X, the autocompletion does not work, as I get 
```
bash: _get_cword: command not found
```

drdanz, 2015-02-20:
@kt10aan Is bash-completion installed? See http://www.linuxask.com/questions/install-bash-completion-for-mac-osx

francesco-romano, 2015-02-20:
I have no problems in using auto completion on OS X

kt10aan, 2015-02-20:
@drdanz , indeed, ``brew install bash-completion`` made it work.

Now there are two ways to continue, one is to source the $YARP_ROOT/scripts/yarp_completion in .bash_profile and the other to link the yarp_completion to `brew --prefix`/etc/bash_completion.d

Both work on my machine, but the first one is a bit cleaner, at least for me that I install YARP from sources.

However, it would be nice to have the YARP formula to install the yarp_completion in `brew --prefix`/etc/bash_completion.d (and possibly add bash-completion as dependency.

kt10aan, 2015-02-23:
I have added the instructions for OS X in http://wiki.icub.org/wiki/Homebrew#Bash_completion.

pattacini, 2015-02-23:
For completeness on the auto-complete feature :), also Windows machines can have it enabled.

@elen4 provided some time ago a way to it along with documentation under `yarp/scripts/WindowsPowerShell`.

fjandrad, 2016-03-07:
I see that in the installation from sources edit ( http://wiki.icub.org/index.php?title=Linux:Installation_from_sources&diff=prev&oldid=20987 ) the suggestion is to use the cp command. I think it would be better to add the suggestion from @drdanz of using the ln:

ln -s $YARP_ROOT/scripts/yarp_completion /etc/bash_completion.d/yarp

Tobias-Fischer, 2016-03-07:
@fjandrad 
This was changed later (http://wiki.icub.org/index.php?title=Linux:Installation_from_sources&diff=21013&oldid=20987) to @drdanz's suggestion. 

However, it got then deleted and replaced with a link (http://wiki.icub.org/index.php?title=Linux:Installation_from_sources&diff=21167&oldid=21077) to http://www.yarp.it/install_yarp_linux.html. Indeed, in this page the bash auto complete is missing.

traversaro, 2016-03-07:
The relevant file for that tutorial is in https://github.com/robotology/yarp/blob/45e06ce13f076efce166711841ab130a04b17d4b/doc/installation/install_linux.dox , so feel free to add the tab completion instructions with a PR. @fjandrad

pattacini, 2016-03-07:
@fjandrad 

You should have received an invitation to join our developers' group.
Once accepted you'll be entitled to assign labels to issues herein.

traversaro, 2017-05-16:
cc @S-Dafarra

francesco-romano, 2017-05-16:
cc @aikolina

closed 2020-12-08 by pattacini