robotology/community #16: Yarp namespace policy

traversaro, 2015-02-20:
In IIT Genoa we have the policy of having different yarp namespaces for the iCub robots (`icub01`, `icub02`, ..) on the different robots. There is any policy on yarp namespace for iCub setups outside Genova?

gsaponaro, 2015-02-20:
In Lisbon we haven't really thought about that, and I am not aware of any official policy.

We only have one full-body iCub robot with namespace "icub" (it was configured like that when delivered from IIT), and a couple of iCub heads with their own laptop/yarpserver/namespace.

Personally, if I were to decide a policy I would make the namespace coincident with ```YARP_ROBOT_NAME``` (if defined) -> one less thing to remember

pattacini, 2015-02-20:
is there any particular reason why we need a policy for that?

lornat75, 2015-02-20:
We do not have a policy. IIT's robot have different namespaces because at some point in the past I think they shared the same network. This temporary solution got cast in stone :) (as it is often the case)

stephane-lallee, 2015-02-20:
Yep in Barcelona we had the same network for two different yarp setups (XIM & iCub). This is great to avoid conflicts, or if you have "students" doing a hands on....

traversaro, 2015-02-20:
No particular reason, we were just cleaning the setup here in paris. Given that there is no convention, we will keep the namespace already in use here, namely to have the namespace coincident with the `YARP_ROBOT_NAME`, that by chance is the same thing @gsaponaro proposed.

closed 2020-12-15 by pattacini