robotology/community #17: Get Raw Values for Calibration Using RobotInterface

lschilli, 2015-02-22:
I recently checked if I can perform finger calibration and arm / arm fine calibration using the the robotInterface. It seems I am missing how to start the robotInterface with a "raw" configuration to be able to read values from the robotMotorGui as described here:
http://robotcub.svn.sourceforge.net/svnroot/robotcub/trunk/iCubPlatform/doc/manuals/ST2009_RBCS_ICUB_ARM_HAND_CALIBRATION_01.doc
(linked from here: http://wiki.icub.org/wiki/Manual)
So the question is: Using the robotInterface, how can the values that need to be entered in the the calibration xls file be determined.

randaz81, 2015-02-22:
Hi, indeed the calibration procedure described in that document refers to the old iCubInterface executable. With robotInterface, the procedure changed a little bit: you don't need anymore to start the robot with a *raw*.ini file. Instead, you just need to edit your general.xml file and set the parameters 'skipCalibration' and 'useRawEncoderData' to true. During the launch of robotInterface, you'll be asked multiple times (one time per part) to press enter and verify that your fault button is pressed. Then you'll be able to take the raw values from the robotMotorGui and put them in the excel file as usual.

lschilli, 2015-03-11:
Thank you. This worked.

closed 2020-12-15 by pattacini