robotology/community #19: Hardware fault in right arm joint 5 is disabling other joints too

kt10aan, 2015-03-10:
As I mentioned in issue #7, our right wrist is stuck. With robotMotorGui, right arm joint 5 is shown as red (with HARDWARE FAULT all over it). Right arm joint 6 is shown as idle.

Now, while the hardware fault is a bad thing, it cannot be helped immediately, but joints 5 and 6 are not needed for what we are doing right now. The rest of the right arm is needed however.

The rest of the arm joints are moving fine with the robotMotorGui. However, in ARE, all right arm actions are not executed (only the hand positions). This is strange as a month ago when I was last using the robot, despite the hardware fault, ARE was executing the right arm actions.

robotInterface (during shutdown) reports that Joint 5 failed to park so it is disabling joint set 3 (which I assume is the whole right arm).

Is there a way to not having the arm disabled (either in ARE or anywhere else) so any actions (in iKin etc) are executed counting ONLY right arm joints 5 and 6 as "disabled"? (E.g. like in my last use of the robot last month, in ARE, with right arm movements working fine ignoring joints 5 and 6). 

And of course, all the above just before a demo tomorrow...so typical :)

barbalberto, 2015-03-10:
To leave those joints disabled, you can change the config file of the right_arm_calibrator from
       <param name="CALIB_ORDER">(0 1 2 3) (4 6 7) (5)</param> 
to
       <param name="CALIB_ORDER">(0 1 2 3) (4 7) </param> 

removing the joint 5 and 6. They will not try to calibrate and willl stay in idle state. 
For the high level control to keep working, it depends the code sending the references is also checking for commands to be executed or not.

pattacini, 2015-03-10:
With the latest major overhaul that has been concerned with the new handling of control modes, we agreed (I and @randaz81) to put extra checks in the code.

The one relevant to your problem I think ca be found [here](https://github.com/robotology/icub-main/blob/master/src/libraries/icubmod/cartesianController/ServerCartesianController.cpp#L1276). Try to comment it out.

kt10aan, 2015-03-10:
@pattacini, should I comment the whole  lines 1276 and 1277 and convert the first ``else if `` to an ``if``?

pattacini, 2015-03-10:
that's right, yep.

kt10aan, 2015-03-10:
Commenting the code did the trick :).
Also, the change in the right_arm_calibrator allows for the robotInterface to exit cleanly (otherwise it wouldn't, needing 4x``Ctrl-C``.
Thanks!

pattacini, 2015-03-10:
cool

closed 2020-12-15 by pattacini