robotology/community #20: [discussion] icub-firmware-shared version catching up with iCub version?

kt10aan, 2015-03-10:
As I had to revert to a previous version of YARP/iCub yesterday, I chose to checkout the tags for the latest package releases (YARP v2.3.63, iCub 1.1.15). Looking for the tags in icub-firmware-shared, the latest one was 1.1.13, which was not working with iCub v1.1.15. I ended up checking out icub-firmware commits from around the date of the iCub v1.1.15 release.

Wouldn't be easier in the future, at the date of each iCub v1.x.y release, to tag the icub-firmware head of the same date as icub-firmware v1.x.y?

lornat75, 2015-03-13:
This looks like a good suggestion to me.

lornat75, 2015-03-13:
@mbrunettini: can you update the tagging instructions so we do not forget next time?

mbrunettini, 2015-03-16:
Fine, tagging instructions updated.

closed 2020-12-08 by pattacini