robotology/community #21: Autocomplete and command history inside "yarp rpc"

gsaponaro, 2015-03-11:
What is required to obtain tab completion and command history support inside a ```yarp rpc /someRpcPort``` terminal session? (The RPC server is a Thrift module responding to queries.)

For example, inside that RPC session I'd like to type ```he<TAB>``` and have it autocompleted to ```help```. And if I press the up key I'd like the command history to show the previous command. I saw that all this is possible on some systems, I wonder why I don't have it enabled. I'm using Ubuntu 14.04 and the latest yarp. Tried a few different terminal programs and window managers.

Note that this is a different matter than https://github.com/robotology/QA/issues/15. That part is working for me (completion of yarp commands in a shell).

pattacini, 2015-03-11:
As far as I know, this behavior depends on your particular OS: e.g. windows has it since long, ubuntu allows for it since quite recent time.

gsaponaro, 2015-03-11:
Thanks.

On our iCub laptop machine where we control all the demos (Ubuntu 14.04 installed from scratch a couple of months ago) we don't have it, but it would really be useful.

On my personal machine (Ubuntu 14.04 upgraded from older versions) I have it.

The difference could be in the configuration of X/terminal/graphics, or in having or not some Ubuntu package which I don't remember anymore. :)

pattacini, 2015-03-11:
I'm afraid I won't be of any help on that...
@alecive, any idea?

paulfitz, 2015-03-12:
@gsaponaro maybe check that the `readline` library is being found when you compile YARP. If you `grep READLINE CMakeCache.txt` you should see something sensible in `READLINE_LIBRARY` and `YARP_USE_READLINE`.

alecive, 2015-03-12:
@gsaponaro I confirm that with the readline library installed (it should be `libreadline-dev`) this behavior works on my machine. For the sake of completeness, this is my `grep READLINE CMakeCache.txt`:

```
alecive@malakim:~/Programmazione/yarp/build (master)$ grep READLINE CMakeCache.txt 
YARP_USE_READLINE:BOOL=ON
//ADVANCED property for variable: YARP_USE_READLINE
YARP_USE_READLINE-ADVANCED:INTERNAL=1
YARP_USE_SYSTEM_READLINE:INTERNAL=ON
```

gsaponaro, 2015-03-12:
Thanks all, ```readline``` it was!

If this feature is not documented (the yarp rpc thrift autocompletion), I suggest to add it somewhere, mentioning the ```readline``` dependency.

I know that the yarp shell autocompletion is already documented in http://wiki.icub.org/wiki/Generic_iCub_machine_installation_instructions

Closing issue.

alecive, 2015-03-12:
@gsaponaro the yarp shell autocompletion is also available here: http://wiki.icub.org/wiki/Linux:Installation_from_sources#Configure_YARP_autocomplete

I will add a couple of lines for the rpc thrift autocompletion then! :)

alecive, 2015-03-12:
Done. It is available here http://wiki.icub.org/wiki/Linux:Installation_from_sources#Compile_YARP

(for the record, I moved it before the cmake configuration because the readline package should be installed before the cmake configuration step. Is it ok @pattacini @paulfitz ?)

pattacini, 2015-03-12:
I believe the commands history is available also for a normal rpc connection (i.e. non-thrift). @alecive, could you update the wiki accordingly?

alecive, 2015-03-12:
Done. :+1:

pattacini, 2016-09-12:
An **importat** update on this.

Since https://github.com/robotology/yarp/pull/892 has been merged, **`libedit-dev`** has replaced `readline` to allow for commands auto-completion, the former package being `LGPL` compatible.

@drdanz may want to add more on this?

Is the documentation up-to-date?

closed 2020-12-08 by pattacini