robotology/community #22: Camera 640x480 is black and white

robertocalandra, 2015-03-14:
Hi all,
We recently started using the eyes/camera at full resolution (640x480 with Bayer encoding). While the left camera worked perfectly, the right camera is now only black and white.
This is quite funny since at low resolution they both work normally (with color).
Do you have any idea of why this could happen and eventually how to fix it?
Thank you.

Tariq-Abuhashim, 2015-03-14:
I used both cameras at the high resolution and they were colored

Tariq-Abuhashim, 2015-03-14:
Check the xml files.  Maybe you are logging the pre-bayerred image

robertocalandra, 2015-03-18:
Hi @Tariq-Abuhashim,
I do not think that this is the case since I am using the same following commands for both left and right eyes.

icubmoddev --from camera/dragonfly2_config_right_bayer_640_480.ini
camCalib --context cameraCalibration --from icubEyes_640x480.ini --group CAMERA_CALIBRATION_RIGHT --name /icub/camcalib/right

With the following config files.
Do you have any other idea?
 
dragonfly2_config_right_bayer_640_480.ini
----------------------------------------------------
device grabber
subdevice dragonfly2raw
width 640
height 480
video_type 3
white_balance 0.477 0.514
gain 0.0
shutter 0.666
name /icub/cam/right
brightness 0
DR2
stamp
sharpness 1.0
hue 0.48
gamma 0.4
saturation 0.271
framerate 30
use_network_time 1
#d 1
guid 00b09d0100b7e8d2
#<64bit global identifier, without the leading 0x> then remove the d option

icubEyes_640x480.ini:
----------------------------------------------------
[CAMERA_CALIBRATION_RIGHT]
projection         pinhole
drawCenterCross    0

w 640
h 480
fx 446.30787465601099
fy 448.44778322512855
cx 313.73434091798771
cy 224.4691342181579
k1 -0.32616218659596552
k2 0.086072743232755614
p1 -0.001835706149853529
p2 -0.001007764102799745

[CAMERA_CALIBRATION_LEFT]
projection         pinhole
drawCenterCross    0

w 640
h 480
fx 443.59549
fy 444.751606
cx 344.89962
cy 207.652054
k1 -0.32121817642921552
k2 0.081499928514089562
p1 -0.000925993397250783
p2 2.6147302026114336e-05

[STEREO_CALIBRATION_CONFIGURATION]
boardWidth 8
boardHeight 6
boardSize 0.09241
numberOfPairs 30

lornat75, 2015-03-18:
Hi Roberto,
this is strange, but i suspect it is a tricky but simple problem. 

Quite unlikely but I recommend you check the camera/dragonfly2_config_left_bayer_640_480.ini to see if there are differences with respect to the "right". In particular check 

width 640
height 480
video_type 3
and DR2

are the important parameters. I would then make sure that the connection uses the bayer carrier (which reconstruct color in the YARP port):

Start the grabbers:

icubmoddev --from camera/dragonfly2_config_right_bayer_640_480.ini
icubmoddev --from camera/dragonfly2_config_left_bayer_640_480.ini

run two viewers, /view/left, /view/right

Connect them:

yarp connect /icub/image/left  /view/left udp+recv.bayer+method.nearest

yarp connect /icub/image/right /view/right udp+recv.bayer+method.nearest

In this way you are baypassing the camcalib completely at least we rule out a connection problem.

lornat75, 2015-03-23:
@robertocalandra?

robertocalandra, 2015-03-23:
Hi @lornat75 ,
Sorry for the delay in answering. I had a look on Friday and 
1) dragonfly2_config_left_bayer_640_480.ini and right have the same:
width 640
height 480
video_type 3
and DR2

2) Trying to connect using the udp+recv.bayer+method.nearest option did not work at all. The viewers were not showing anything.

Suggestions?

gsaponaro, 2015-03-23:
Hi @robertocalandra, maybe this is obvious, but on the machine where you are running ```yarpview```, can you confirm that the packages ```libraw1394-dev libdc1394-22-dev``` are installed, and that yarp is compiled with the following options?
````
CREATE_OPTIONAL_CARRIERS
ENABLE_yarpcar_bayer_carrier
````

robertocalandra, 2015-03-23:
Hi @gsaponaro. 
At the moment I cannot confirm, but even then, why would only one of the camer be in black and white and the other one with color?
I will check later about the packages.

gsaponaro, 2015-03-23:
You're right, the issues shouldn't be related.

I was replying in response to "2) Trying to connect using the udp+recv.bayer+method.nearest option did not work at all. The viewers were not showing anything."

Another possibility is to try ```mcast+recv.bayer+method.nearest``` instead of ```udp+recv.bayer+method.nearest```. In the past we tried both and ended up using ```udp```.

lornat75, 2015-03-23:
@robertocalandra:

> Trying to connect using the udp+recv.bayer+method.nearest option did not work at all. The viewers were not showing anything

You mean both viewers (right and left) show black? Any output from the yarp connect command? If both left and right cameras do not work there could be an issue similar to what @gsaponaro reports. Otherwise we should investigate why right and left camers behave differently. Please use tcp just to be on the safe side it is usually less problematic than udp and mcast.

lornat75, 2015-03-23:
@robertocalandra, let me elaborate a bit more.

If at least one of the cameras show correct color information it means that the computer that runs the camcalib has the libraries performing bayer interpolation (the ones that @gsaponaro mentions). If you run the viewers on that same machine you should be able to see color at least in one of the viewers. Can you please do this?

robertocalandra, 2015-04-07:
Solved. It turned out that it was a strange combination of concause. And @gsaponaro was right that the bayer carrier was not enabled!

closed 2020-12-15 by pattacini