robotology/community #23: Gravity and friction for the simulated iCub robot.

hixboty, 2015-03-23:
Hi, i want to test some walking patterns inside the simulator and even if i move the limbs in a very unnatural position, iCub is still standing instead of falling. I know there is a simple way of creating a ball that is affected by gravity inside the simulator... but can i turn on the effect of gravity for the simulated iCub robot? I know that the limbs have mass so i guess that the effect of gravity is turned off for it.

Is there currently a way of achieving this inside the simulator?

Thank you,
Cristian

pattacini, 2015-03-23:
Try to disable the parameter `fixed_hip` contained in the `iCub_parts_activation.ini` file - the one actually used by the simulator. To know while file is the one, do launch `yarp resource --context simConfig --file iCub_parts_activation.ini`.

hixboty, 2015-03-23:
It works like a charm. The robot simulation experiences gravity and friction by disabling the  fixed_hip parameter, and as a bonus.. if it falls too hard it will break into pieces :) The simulator does crashes when the robot breaks but i'll see what i can do. Maybe i'll make it start from a "lay down on its back" position. Thank you very much :)

closed 2020-12-09 by pattacini