robotology/community #24: announcement: YARP docker virtual image available

giodegas, 2015-03-24:
Dear iCub-ists,

I just published a YARP docker image which allows to install a YARP server in a cloud computing environment based on docker application virtualization:
https://registry.hub.docker.com/u/giodegas/yarp/

The docker file lives in this github repository:
https://github.com/giodegas/docker-yarp/tree/binary

to install as a new YARP image on docker:

    docker pull giodegas/yarp

to start a shell:

    docker run -it giodegas/yarp bash
    
to check installation:

    yarp check
    
to start the server:

    yarpserver
    
then open a browser session to http://localhost:10000

Beta testers are welcome to improve it.

Regards
Giovanni De Gasperis

pattacini, 2015-03-24:
Thanks @giodegas for contributing!

giodegas, 2015-04-02:
I just committed an update with better instructions on the [ReadME](https://github.com/giodegas/docker-yarp/blob/master/README.md) file, so to have the server running on a non routable IP (192.168.0.1)

closed 2020-12-12 by pattacini