robotology/community #25: world ball beginning position

matusko89, 2015-03-26:
Hi all,
I had somehow managed to overwrite the build in ball starting position and now it is positioned under the table and I have to set it position every time I start the simulator. I tried to find something in the simConfig directory but with any luck. I assume there should be some .ini file for that but I am unable to find it.

Thanks 
Matus

vtikha, 2015-03-27:
Hi @matusko89, the short answer is that, at the moment, the default sphere position cannot be configured in the simConfig or in any ini file. You will have to set its position or initialise the simulator's world as you please from your module, once you start the simulator. This has been noted as a future feature request for the simulator.

lornat75, 2015-03-27:
@matusko89: how did you manage to change the ball position? in the code? Maybe you need to reset your repository and discard your local changes:

git reset --hard

be careful because you lose all your local changes.

vtikha, 2015-03-27:
@lornat75 I think this is related to this issue #18

matusko89, 2015-03-31:
@lornat75 I was messing around with yarp rpc /icubSim/world and trying if it is maybe possible to change color of ball. I managed to crash the simulator few times and the ball just happened to start render under the table...

matusko89, 2015-03-31:
Nevertheless It is back to normal now but some .ini file would be nice for future.

Thanks

vtikha, 2015-03-31:
@matusko89, make sure that you are not trying to place the ball within the table as that might create some issues due to various collisions. I should suggest to use the custom object creation instead of the default ones as you have more control on their properties. You can find everything [here](http://eris.liralab.it/wiki/Simulator_README#OBJECT_MANIPULATION:)

closed 2020-12-09 by pattacini