robotology/community #26: robotInterface creates a lot of threads and some of them fails with error `A thread failed to start with error code:`

traversaro, 2015-04-08:
Programs that use Yarp tend to use a lot of threads. 

In particular using the `robotInterface` on the iCub pc104 it can happens that a thread fails to start with the following message: `A thread failed to start with error code -1`, obviously creating big problems in the use of the `robotInterface`.

traversaro, 2015-04-08:
Usually this happens because the huge number of threads spawned by the `robotInterface` (for each new connection new threads are spawned) are saturating the memory. This because every new thread takes a fixed amount of memory for its stack. 

On typical Debian system, every thread by default takes 8192 KiloBytes for its stack. In IIT we decreased the default stack size for thread using the `ulimit` utility. In particular, you can decrease the default stack size to 4096 KB with:
~~~
ulimit -s 4096
~~~
In this way you implicitly increase the number of threads that is possible to create in the system. 
Anyway, several other strategies can be used to reduce the number of threads created by the robotInterface, for example by using `multicast` connection instead of point to point connections. 

However consider that `ulimit -s 4096` only modifies the default stack size in the current session.

traversaro, 2015-04-08:
A more permanent and proper way of reducing the stack size seems to be editing the `/etc/security/limits.conf` file.

cc @lornat75 @apaikan @mbrunettini @barbalberto  It could make sense to modify the default pc104 image to come with a default stack size of 4096 ?

lornat75, 2015-04-08:
Consider that the new Linux image is 64bits and does not have this limitation.

In addition: many threads are created by remote instantiation of controlboards; if I remember correctly many are just created to access encoders. We should create a minimal controlboard (or re-use the same with a parameter --status_only) that creates only the port to read encoders/status.

paulfitz, 2015-04-09:
There's also the possibility of having a controlboard just work with a single port rather than 3. There used to be a flag for that (`--single_port`).

traversaro, 2015-04-14:
cc @randaz81 @jeljaik

closed 2020-12-15 by pattacini