robotology/community #27: Mismatch in real and measured velocities at the wrist pronation-supination

iron76, 2015-04-09:
There seems to be something wrong with the measured velocities at the wrist pronation-supination (joint 4) in the iCubParis02. In particular, if I use the robotMotorGui to control a movement 0[deg]->90[deg] and I set the (average) velocity set to 10 [deg/s] the measured instantaneous velocity is always below 1[deg/s] which seems quite unlikely. For other joints this is not the case.

To measure the velocity I am using the controlBoardDumper properly configured. @randaz81 do you have any idea if I am doing wrong or if something is mis-configured?

randaz81, 2015-04-09:
Hi @iron76,you are doing right. 
Looking at iCubParis02 configuration files (motorControl\icub_left_arm.xml), there is a section:

`<group name="VELOCITY">      `
`<param name="MOT_speed_estimation">       1  `
`<param name="MOT_accel_estimation">       1  `
`<param name="JNT_accel_estimation">       5  `
`<param name="JNT_speed_estimation">       5  `

However, the MC4 board does not implement in the firmware the moving average filter requested by these parameters (which are instead implemented by BLL boards), so the resulting speed is wrong.
As temporary fix, you can set all the above parameters to 0, and you should get a consistent measurement.
A further improvement (available in the next days) is to implement the filter also on the MC4 board. In this way you should get less noisy measurements. 
I'll notify you as soon as the fw fix is ready.

pattacini, 2015-06-23:
Stale issue, closing

closed 2020-12-15 by pattacini