robotology/community #29: I cannot perform FW upgrade anymore from icub-firmware/build

marcoaccame, 2015-05-11:
Hi everybody, on my iCubGenova04 I have just updated the svn repository icub-firmware and ... the /build folder has just disappeared. The problem now is that I don't know how to perform FW update.

Is there anybody who can help me?

traversaro, 2015-05-11:
I guess you should use the new https://github.com/robotology/icub-firmware-build repo?

marcoaccame, 2015-05-11:
Hi,
I have seen that on the svn repository <icub-firmware> everything has disappeared apart from the file called "we.have.moved.to.github.readme.txt".

That means that the svn repository you are using is not made obsolete.

If you read the file, you have instructions of how to retrieve on your iCub the binary files that you need to perform FW update.

The binaries have moved into the following github repository:  https://github.com/robotology/icub-firmware-build

All you just need to do is:

1. remove folder icub-firmware, or rename it (so that any other user of the robot knows it is obsolete)
2. retrieve the new repository with commend "git clone https://github.com/robotology/icub-firmware-build".

Inside folder icub-firmware-build/ there are the binaries that you need.

marcoaccame, 2015-05-11:
Yes, Silvio indeed.

The old svn repo is now obsolete and has been split into two new git repositories:
1. one is https://github.com/robotology/icub-firmware-build and contains whatever is needed on iCub;
2. https://github.com/robotology/icub-firmware and contains the source code, which is not used by iCub.

On iCubGenova04 icub-firmware-build is already installed. I wrote this Q&A so that people using other robots can solve in case they make an update to the svn repository and lose their data.

traversaro, 2015-05-11:
Sorry, I did not get it was a fictitious question. :D 
For properly notify that the procedure for installing the firmware has changed we should updated the Firmware wiki documentation at http://wiki.icub.org/wiki/Firmware (that I guess it's the first hit on google for the search "icub firmware").

closed 2020-12-15 by pattacini