robotology/community #31: thrift client implementation

tanismar, 2015-05-17:
Hi, 
I have a module 'A' which defines a thrift server in order to receive rpc commands. One of those calls shall return a user defined struct (a 3D point with .x .y .z fields). 
Another module 'B' from a different repository needs to call that function and read the returned structure. I am trying to figure out which would the best way to make them communicate, but I haven't yet come to a suitable answer: On the one hand, I could implement a thrift client in module B, but given that it belongs to a different build from A, the thrift includes (with the server and client implementations) are not available (or I don't know how to include them). On the other hand, I could set a normal rpc port communication. In that case, the point3D structure would be returned within the reply Bottle. Should it be read with .asList()? 
Which is the best option for this case? 
Thanks in advance!

lornat75, 2015-05-18:
A possible solution without sharing thrift files is to implement and export in the module A a small library that defines the required types  and contains the client functions (basically the client side of the code generated by thrift).

francesco-romano, 2015-05-18:
We are trying to do what @lornat75 suggested in the `clock` plugin in https://github.com/robotology/gazebo-yarp-plugins.
Basically we created a library which contains only the autogenerated code.
This library is linked by the "server" directly in the application.
It is also installed and properly exported through cmake (this last part still have to be done), so that clients can simply look for it with `find_package`.

traversaro, 2015-05-19:
`YCM` (the package containg several CMake macro developed at IIT) can be useful for correctly exporting C++ libraries, even libraries that are automatically generated from `thrift`. 

A tutorial on how to do this is available at: http://robotology.github.io/ycm/gh-pages/master/manual/ycm-build-system-support.7.html . 

Some example repositories are also available:
* https://github.com/tacman-fp7/template-pkg
* https://github.com/robotology-playground/lib-template-cmake

drdanz, 2015-05-26:
Another option could be to install the thrift files on the server side and to use them and regenerate the thrift includes on the client side...

closed 2020-12-08 by pattacini