robotology/community #32: Mjpeg carrier... how to use?

lukeboorman, 2015-05-20:
Hi. We would like to compress the iCub image streams using the Mjpeg carrier in YARP but are struggling to use this. We will stream the camera output to a remote location and so need to reduce bandwidth usage. Is there an example on how to do this?

I have turned on the mjpeg carrier in YARP cmake and have the autocompress on, is there a setting we are missing in the icubmoddev frame grabber? The Mjpeg_test executable I assume only decompresses the stream?

I have started using the mjpegstream c++ class, but the include file isn't moved across with the yarp install and I assume I don't need to use this?

Ps How do I add labels #Discussion #Software

drdanz, 2015-05-20:
Hello @lukeboorman, the mjpeg_carrier must be compiled in order to work, please check that the `ENABLE_yarpcar_mjpeg_carrier` flag is enabled when building yarp on both sides.

The easiest way to use the mjpeg carrier is this on the same pc/network:

```
(shell 0) yarp server
(shell 1) yarpdev --device test_grabber --name /grabber
(shell 2) yarpview --name /view
(shell 3) yarp connect /grabber /view mjpeg
```

Another way to use it, is to connect a stream generated by yarp to an external application

```
(pc1) yarpdev --device test_grabber --name /grabber # yarpserver must be running
   # get the ip address and port from the output (here I'm assuming 192.168.99.1:10002)
(pc2) open browser at http://192.168.99.1:10002/?action=stream
```

To connect 2 different yarp ports running on different networks, you should be able to do something like this:

```
(pc1) yarpdev --device test_grabber --name /grabber # yarpserver must be running
   # get the ip address and port from the output (here I'm assuming 192.168.99.1:10002)
(pc2) yarpview --name /view # yarpserver must be running, but not necessarily the same instance
(pc2) yarp connect http://192.168.99.1:10002 /view mjpeg
```

Finally the mjpeg carrier allows you to connect to any external mjpeg stream you can reach, you will have to do something like this (to be honest I never tried it and I was not able to do it right now, perhaps @apaikan can add something):

```
yarp name register /streamer tcp ... 8080
yarpview /view
yarp connect /view /streamer mjpeg # Note that the connect order because this is a "pull" protocol
```

lukeboorman, 2015-05-21:
Thanks for the info, will give the test_grabber a go today.

apaikan, 2015-05-21:
@lukeboorman the instruction provided by @drdanz should work. 
The `mjpeg` carrier automatically compress and decompress  the images and you do not need to implement anything to get it work.  If you have compiled Yarp with `ENABLE_yarpcar_mjpeg_carrier=ON`, it is enough to connect two yarp ports using the `mjpeg` carrier: 
```
yarp connect /sender /receiver mjpeg
```

`mjpeg` carrier also allows you to stream out Yarp images and visualize them using any standard application (e.g., mplayer, firefox). For example to see a Yarp image from a web browser (can be a browser from your mobile phone too) follow these steps: 

- run a `test_grabber`
```
$ yarpdev --device test_grabber 
[INFO]Test grabber period 0.0333333 / freq 30 , mode [line]
[INFO]created device <test_grabber>. See C++ class yarp::dev::TestFrameGrabber for documentation.
yarp: Port /grabber active at tcp://10.1.89.119:10002
yarp: No framerate specified, polling the device
[INFO]created wrapper <grabber>. See C++ class yarp::dev::ServerFrameGrabber for documentation.
yarp: Port /test_grabber/quit active at tcp://10.1.89.119:10003
[INFO]device active...
[INFO]device active...
```

-  open a firefox/iceweasel browser (chrome does not support mjpeg streaming) and enter the ip address of the test_grabber in this format: 

```
http://10.1.89.119:10002/?action=stream
```
Let me know if you encounter any problem. 

P.S. You can read more on that here : http://apaikan.net/papers/Fitzpatrick14middleway.pdf

lukeboorman, 2015-05-21:
Thanks we have got this working between the icub laptop and another ubuntu pc uisng yarpview.... "yarp connect /icub/cam/left /yarpImgView mjpeg".

However we are having some issues:

1. I can't get this to run on a windows machine (probably my config!), Running the yarp connect... it finds the mjpeg carrier, but says "yarp: Failed to find library support for carrier mjpeg". The PC has yarp_data_dirs etc setup and the mjpeg_carrier.dll have been produced and is in the 'install' location.
 
Other comments:
1. When closing the connection we have to use "yarp disconnect /icub/cam/left mjpeg://yarpImgView" otherwise it says the connection doesn't exist.
2. On connecting the return message from yarp seems to be the wrong way round (to -> from), or is this a two-way link? "Success: Added connection from /yarpImgView mjpeg://icub/cam/left

Thanks again for the help

drdanz, 2015-05-21:
There are some issues on windows, I managed to get it working, but I don't remember exactly what I had to change, and also it works only in some cases, I will investigate this a little bit more and try to fix it...

Tobias-Fischer, 2015-05-21:
Hi @lukeboorman,
I can only help with your "other comment" #2: See the comment from @drdanz, where it is written that mjpeg is a pull protocol. Therefore, sender and receiver are the wrong way around. Did you try the same trick with other comment #1?

pattacini, 2015-06-12:
@lukeboorman any update?

lukeboorman, 2015-06-15:
Hi, thanks all! We got MJPEG stream running and used gateways to send the data over a different network. Much increase in frame rate and we were able to increase the resolution using Mjpeg with bayer, which overcomes some of the visible effects of the compression. I still struggle to run on my windows without running from within the yarp build directory, I think I have all the environment variables set up correctly.

pattacini, 2015-06-18:
... hence it should suffice to have `yarp/build/bin/release` in the path, right?

closed 2020-12-08 by pattacini