robotology/community #35: reading a YARP port from ROS

nigno17, 2015-06-02:
Hi everybody,
I'm trying to control the iCub robot in Gazebo using ROS and the yarp gazebo plugin. 

My goal is to read the encoders and send motor commands to the joints from ROS. I was following this tutorial: http://wiki.icub.org/yarpdoc/yarp_with_ros.html , but I don't understand how can I read an existing yarp port from ROS. Does it exist an automatic way to connect a yarp port to a ros topic (something like "yarp connect 'icubGazeboSim/head/state:o topic://head_enc") or I have to write a yarp program that reads from the port and publishes on the topic by hand?

apaikan, 2015-06-03:
@nigno17 

There is an example in the `example/yarptos_examples` folder of the Yarp repository called `grab_encodrs.cpp` which I guess does the same thing you need. Have you tried that? 

Here is the code according to the example: 

Ros message : 
```bash
int32 v_tag # set to 266 (BOTTLE_TAG_LIST+code)
float64[] v # suggested length: 16
```

and the C++ Code: 

```C++
#include <ros/ros.h>
#include <yarpros_examples/Encoders.h>
#include <stdio.h>

void chatterCallback(const yarpros_examples::EncodersConstPtr& enc)
{
  printf("Encoder readings for first three joints: %g %g %g\n",
	 enc->v[0], enc->v[1], enc->v[2]);
}

int main(int argc, char** argv)
{
  ros::init(argc, argv, "yarp_encoder_listener");
  ros::NodeHandle n;
  ros::Subscriber chatter_sub = n.subscribe("pos", 1, chatterCallback);
  ros::spin();
}
```

Tobias-Fischer, 2015-06-03:
Did you also see this: http://sourceforge.net/p/robotcub/mailman/message/33947143/ ?

nigno17, 2015-06-03:
Thanks a lot. I'm using the wrapper and everything is working.

pattacini, 2015-06-03:
@nigno17 can we close this issue then?

nigno17, 2015-06-03:
yes thanks!

closed 2020-12-08 by pattacini