robotology/community #36: Relation of pages/rows in iCub_Calibration_*.xls with xml calibration files

lschilli, 2015-06-03:
A while ago I performed an arm calibration. It took me longer than usual to figure out how the calibration files need to be updated. Previously the files generated by the xls document where directly used by the iCubInterface. For the robotInterface I am looking for a one to one mapping that explains where in which xml section each part of the xls output goes.
For example:
```
sheet "left_arm" rows 3,4 --> hardware/motorControl/icub_left_arm.xml /devices/group[name=="general"]/.../param/....
sheet "left_arm" rows 23,26 --> calibrators/left_arm_calib.xml /group[name=="HOME"]...
```
(note: xpath meant to be accurate in this example)
If I am not wrong different parts of the same sheet go to different files now, so this mapping would be extremely helpful to quicker understand the structure. For the finger calibration I have not fully understood it yet.

So the question is: Is such a mapping of excel generated ini file to robotInterface xml file already somewhere documented? Or are other changes in generating calibration files planned / I was not aware of them?

pattacini, 2015-06-18:
Is anybody of @robotology/icub-support-admin or @robotology/icub-developers willing to help @lschilli on this topic?

Do we have some documentation on that?

pattacini, 2015-06-23:
closing as **unanswered**.

closed 2020-12-15 by pattacini