robotology/community #39: Confirming arm version number of our iCub

gsaponaro, 2015-06-17:
Hi, in short we would like to know if the arms mounted on iCubLisboa01 are v1 or v1.7, with reference to http://wiki.icub.org/wiki/ICubFowardKinematics_right (there is a small change in one DH parameter between the two versions).

When we launch the software module ```iKinCartesianSolver --part right_arm```, it prints the DH parameters of the arm v1 as follows, therefore we believe that we have an arm v1, can you confirm it?

DH Table:  (H0 (0.0 -1.0 0.0 0.0 0.0 0.0 -1.0 0.0 1.0 0.0 0.0 0.0 0.0 0.0 0.0 1.0)) (type right) (numLinks 10) (HN (1.0 0.0 0.0 0.0 0.0 1.0 0.0 0.0 0.0 0.0 1.0 0.0 0.0 0.0 0.0 1.0)) (link_4 (A 0.0) (D 0.0) (alpha -90.0) (offset -90.0) (min 0.0) (max 160.8)) (link_3 (A 0.0) (D -0.10774) (alpha 90.0) (offset -90.0) (min -95.5) (max 5.0)) (link_7 (A 0.0) **(D -0.1373)** (alpha 90.0) (offset -90.0) (min -50.0) (max 50.0)) (link_5 (A -0.015) (D -0.15228) (alpha -90.0) (offset -105.0) (min -37.0) (max 100.0)) (link_6 (A 0.015) (D 0.0) (alpha 90.0) (offset 0.0) (min 5.5) (max 106.0)) (link_2 (A -0.023365) (D -0.1433) (alpha 90.0) (offset -105.0) (min -59.0) (max 59.0) (blocked 0.0)) (link_0 (A 0.032) (D 0.0) (alpha 90.0) (offset 0.0) (min -22.0) (max 84.0) (blocked 0.0)) (link_1 (A 0.0) (D -0.0055) (alpha 90.0) (offset -90.0) (min -39.0) (max 39.0) (blocked 0.0)) (link_9 (A 0.0625) (D 0.016) (alpha 0.0) (offset 180.0) (min -25.0) (max 25.0)) (link_8 (A 0.0) (D 0.0) (alpha 90.0) (offset 90.0) (min -65.0) (max 10.0)) 

By the way, this robot is an iCub v1.2.1, according to http://wiki.icub.org/wiki/ICub_versions

/cc @vicentepedro @plinioMoreno

pattacini, 2015-06-18:
Hi @gsaponaro 

Looking at the `iCubLisboa01` conf files I can confirm your intuition is right.
Btw, @robotology/icub-support-admin can say the last word on it.

pattacini, 2015-06-18:
closing, feel free to reopen it for further info

gsaponaro, 2015-06-23:
Thanks, so it's as we thought, but it would be perfect if we had the confirmation from the support team as well (based on the CAD models, your records of our robot after our arms + hands upgrades, etc.).

@vicentepedro would like to have this information to compute the Robot Jacobians, see https://github.com/robotology/QA/issues/38

/cc @robotology/icub-support-admin @robotology/qa-users

pattacini, 2015-06-23:
Which hand upgrades are you referring to, in particular?
I've just double checked with our maintenance here in the lab and you have arms v. 1.0.

gsaponaro, 2015-06-23:
For example, during the July-September 2011 hardware upgrades of our robot at @ IIT, I was not sure whether the arms had been changed. Probably not. I remember that F/T and touch sensors were installed in that occasion. 

Thanks for the double check! Closing. :+1:

pattacini, 2015-06-23:
2011 wow, it's a life ago
pretty sure the hand v2.0 were still _in mente Dei_

closed 2020-12-15 by pattacini