robotology/community #40: miscallenous questions after a first contact with yarp-gazebo

gavril0, 2015-06-17:
I have recently decided to look at a robot simulator that could be potentially connected to a NN simulator.  For a variety of reasons, one is being at IIT, I have decided to use gazebo-yarp-plugin with simbody as robot simulator. 

After installing  [Simbody from sources ] ( https://simtk.org/docman/view.php/47/1583/HowToBuildSimbodyFromSource_MacLinux.pdf ),  [ Gazebo using Ubuntu packages ] ( http://gazebosim.org/tutorials?tut=install_ubuntu&ver=5.0&cat=install ),  [ YARP  from sources] ( http://wiki.icub.org/wiki/Linux:Installation_from_sources#Getting_the_YARP_and_iCub_sources ),  the [Gazebo-Yarp-Plugin ] ( https://github.com/robotology/gazebo-yarp-plugins ) and the [ iCub gazebo models ] ( https//github.com/robotology-playground/icub-gazebo),  I was able to simulate the double_pendulum model in gazebo and to control it with yarpmotorgui :-). 

Since this is basically my first contact with Yarp, Gazebo and even Ubuntu (I have been programming under Windows for years), I decided to learn how to use all t at with  [iCub tutorial ] (http://wiki.icub.org/iCub/main/dox/html/icub_tutorials.html).  I was also able to adapt [icub arm tutorial ] (http://wiki.icub.org/iCub/main/dox/html/tutorial__arm_8cpp.html) to control the position of the [double pendulum model] (https://github.com/robotology-playground/icub-gazebo/tree/master/double_pendulum) with my program through yarp. 

I found a lot of useful information online but I was also left with some questions. Sorry for this rather long and mixed bag of questions but my colleagues at IIT suggested that I post them here as their responses could be profitable to other users.  

1) how to avoid running yarpserver and gazebo as superuser ? (all installation was done with sudo -s)

2) In my installation, the file icub-gazebo/double_pendulum/model.urdf contains only the following line:

> model://double_pendulum/conf/gazebo_controlboard.ini 0.0 0.0 

This is different from the [model.urdf] (https://github.com/robotology-playground/icub-gazebo/blob/master/double_pendulum/model.urdf) on github. I don't understand why it is different and how gazebo finds the pendulum model in my installation. 

3) In the [gazebo_controlboard.ini] (https://github.com/robotology-playground/icub-gazebo/blob/master/double_pendulum/conf/gazebo_controlboard.ini) file, where/how is defined the interface (IPositionControl, IVelocityControl, ....) actually used by the control board in the simulation? Where do I get more info
on the format of this file (http://wiki.icub.org/yarpdoc/group__controlboard__basic.html is also very limited)

4) From a program, how do I know which interface (IPositionControl vs IVelocityControl vs IImpedanceControl) is actually controlling the robot?  I guess that they are mutually exclusive and that you cannot use IVelocityControl interface on a "remote_controlboard" device if the "controlboard" device to which it is connected use a different interface. How do I switch from one interface to another one?

5)  I found little guidance about how/when to use the different motor interfaces listed in this [page] (http://wiki.icub.org/yarpdoc/group__dev__iface__motor.html). For example, what is the actual difference between Position control with PID and impedance control (besides the API) ? Is there any actual difference in the EMS board firmware ?

6) I am not sure to fully understand the wall_clock mechanism implemented in the yarp-gazebo-plugin. I understood that any control thread will use the wall_clock to call its callback functions but is the synchronization between Gazebo and control module two ways ?  If the control module is slower than the Gazebo simulation, is there a way to stop gazebo until the controller has computed and sent the commands ?

7) What IDE to use (K-develop ? ) and how to set it up to be able to browse yarp source code ? 
	(this would be useful to have an IDE to be able to navigate into yarp source code)    

P.S.  (gavril0 = Gabriel Baud-Bovy)

traversaro, 2015-06-17:
Hi @gavril0 , thanks for taking the time to post on QA !

I will try to reply inline: 

> 1) how to avoid running yarpserver and gazebo as superuser ? (all installation was done with sudo -s)

In general you should not need to run program using sudo, even if you installed it as superuser. 
To check that your permission are correctly set, can you post the output of the following commands:
~~~
 ls -l `which yarpserver`
~~~
 and
~~~
ls -l `which gazebo`
~~~

> 2) In my installation, the file icub-gazebo/double_pendulum/model.urdf contains only the following line:
> model://double_pendulum/conf/gazebo_controlboard.ini 0.0 0.0
> This is different from the [...]

That is definitely strange.. Are you sure that Gazebo is actually loading that urdf file? What is the value of your `GAZEBO_MODEL_PATH` enviromental variable. 

> 3) In the gazebo_controlboard.ini file, where/how is defined the interface (IPositionControl, IVelocityControl, ....) actually used by the control board in the simulation? Where do I get more info
on the format of this file (http://wiki.icub.org/yarpdoc/group__controlboard__basic.html is also very limited)

The different control modes are not selected at configuration time, but at runtime using the [IControlMode2](http://wiki.icub.org/yarpdoc/classyarp_1_1dev_1_1IControlMode2.html) interface. For more information on control modes, check this file : http://wiki.icub.org/images/c/cf/ICub_Control_Modes_1_1.pdf  . For an example on how to switch control modes, check [this tutorial](http://wiki.icub.org/brain/icub_impedance_control_tutorial.html).

For more information on the parameters of the controlboardwrapper, please check http://wiki.icub.org/yarpdoc/classyarp_1_1dev_1_1ControlBoardWrapper.html . Documentation for gazebo devices is still a work in progress, that you can track in this issue: https://github.com/robotology/gazebo-yarp-plugins/issues/95 (first items). 

> 4) From a program, how do I know which interface (IPositionControl vs IVelocityControl vs IImpedanceControl) is actually controlling the robot? I guess that they are mutually exclusive and that you [...]

You can check the `IControlMode2` interface, and for further information this slides : http://wiki.icub.org/wiki/File:Icub_control_modes_slides_1_1.pdf . 

> 5) I found little guidance about how/when to use the different motor interfaces listed in this page. For example, what is the actual difference between Position control with PID and impedance control (besides the API) ? Is there any actual difference in the EMS board firmware ?

You can find this information in the control mode documentation available in  http://wiki.icub.org/wiki/File:Icub_control_modes_slides_1_1.pdf and http://wiki.icub.org/images/c/cf/ICub_Control_Modes_1_1.pdf  . In a nutshell, in the position control the output of the low level joint position trajectory generator is given as input to a PID (with a feedback in position) that is controlling directly the motor input. In impedance control, the output of the low level joint position trajectory generator is given as input to a PD control law with a feedback in position (impedance). The output of this PD is then considered a desired torque value, that is then sent to a lower level PID (with a feedback in torque from force sensing) that has to track the desired torque. 

> 7) What IDE to use (K-develop ? ) and how to set it up to be able to browse yarp source code ?
(this would be useful to have an IDE to be able to navigate into yarp source code) 

If you are on Linux, the IDE that directly support CMake (KDevelop or QtCreator) can be the best choice for quickly opening CMake-based projects. To open the Yarp source code in KDevelop: Open KDevelop --> Click on "Project" Menu --> Click on "Open/Import Project" ---> Open the CMakeLists.txt in the top level directory of Yarp repository . 

The synchronization system of Gazebo is a bit tricky and not so clear even to me, so I will reply tomorrow on that.

gavril0, 2015-06-18:
@traversaro. Thanks for the info. Still interested by the question on synchronization as there will be a similar issue for the NN simulator. Planning to look at [iSpike] (http://ispike.sourceforge.net/).  I'll digest the pointers you gave me and come back here for further questions if needed (doing it a bit on my spare time at the moment).

pattacini, 2015-06-28:
@gavril0, closing, feel free to reopen it.

closed 2020-12-08 by pattacini