robotology/community #42: Doubts about Configuration Files in YARP

pattacini, 2015-07-02:
I'm a bit lost among all the possible paths from where a configuration file
can be found and loaded by a YARP module.

I've got a bunch of questions:

1. Where can I read more about the policies used by YARP to locate files on the system?
2. Is there an handy way to easily find out which exact file will be loaded at start up,
given that there exist several copies of it in different paths?
3. What do I have to do when I want to modify the content of a configuration file?
4. What do I have to do when I want to make those modifications become the default values?

pattacini, 2015-07-02:
The way YARP locates files by means of the [`yarp::os::ResourceFinder`](http://wiki.icub.org/yarpdoc/classyarp_1_1os_1_1ResourceFinder.html) class may appear not easy to
understand at first sight, but stems from the original goal of **shipping binaries alongside their configuration files** with the need to **accommodate to different platforms** (_Windows_, _Linux_, _Mac_) and to **different credentials** a user might own on the particular system (i.e. he may not have writing access to system paths).

To answer the questions:

1. The documentation is quite extensive on this topic and can be browsed here: http://wiki.icub.org/wiki/Configuration_files. This page is linked from the [Manual](http://wiki.icub.org/wiki/Manual).

2. Whenever you find yourself uncertain about which file your module will load at start up, just use one of the following utilities:

  a. `yarp-config context --where moduleContext`: this command will list down all the context paths containing you configuration file as they will be visited during the search in decreasing priority order.
  b. `yarp resource --context moduleContext --find fileName.ini`: this command will show you the location of the file "_fileName.ini_" belonging to the context "_moduleContext_" as result from the search. All the possible paths will be thus visited according to the YARP policies.

3. You should first copy the file in the local path where it is sure you have writing access (e.g. in `~/.local/share/yarp/context` in _Linux_): to automate this process, you can rely on this command: `yarp-config context --import moduleContext filenName.ini`. This local file will "**shadow**" any other copy in the file system.
4. You should overwrite the copy of the file in your repository and then commit it to the net. Afterwards, you have to issue a "**cmake**", because only by "cmaking" the project, that file will be copied out in the build path where it will be ready to be loaded. Possibly, you have to remember to "**make install**" the project too if you did in the past. At this point you can remove the file in the local path by: `yarp-config context --remove moduleContext fileName.ini`.

pattacini, 2015-07-02:
/cc @robotology/icub-support-admin and anyone else has doubts on this
feel free to expand this issue.

traversaro, 2015-07-02:
/cc @robotology/codyco-developers

drdanz, 2015-07-02:
Scary details about ResourceFinder internals: http://wiki.icub.org/wiki/YARP_ResourceFinder

pattacini, 2015-10-14:
There's an important update on the use of `yarp-config robot` utility. Importing single files of the robot context is no longer allowed. One can only decide to import in the local path the whole set of `xml` files.

@randaz81 and @drdanz might provide rational for this change.

One question, then, about importing single `.ini` file in the robot context. Is it still allowed? We do use `.ini` files in robot contexts to better customize applications for specific robots, indeed.

You ensured that the `ResourceFinder` still works as expected on the `.ini` files in those robot directories, but is `yarp-config robot --import robotName fileName.ini` still operative?

drdanz, 2015-10-16:
@pattacini No, since robotology/yarp@560ac7214a746243351daf6972b3300136eb54e8 you are only allowed to import all the files from a robot.

The reason is that for a few good reasons (i.e. being able to modify the robot files with external xml editors), we are moving the robotInterface to use the [XInclude](http://www.w3.org/TR/xinclude/) W3C Recommendation instead of our custom include tags. This requires that all the files for the robotInterface are in the same place, and the ResourceFinder can no longer be used to look for included files.

In order to ensure that all the robotInterface files are imported together, we decided to force the users to import *all* the files for a robot. We might consider to restore this functionality in the future, if this is a very requested feature, but as a matter of fact at the moment it is disabled.

randaz81, 2015-10-16:
@pattacini  you can always use `yarp-config robot --diff robotName` to check the status of individual files.

pattacini, 2016-02-03:
@tanismar and @GiuliaP please take a glance at these notes for dealing with configuration files in yarp.

closed 2020-12-08 by pattacini