robotology/community #44: Red ball demo without impedance

kt10aan, 2015-07-14:
Hello,

I think that, the red ball demo had two versions: with impedance and without.

But now, only the impedance version seems to be around. However, the impedance mode in iCubOsaka01 is not working properly right now (to be investigated). If I switch the ``impedance`` to ``off`` in the config files of the module, in icus-basic-demos, would it produce the non-impedance enabled behaviour of the demo?

pattacini, 2015-07-14:
Hi @kt10aan,

The so-called `red ball demo` has still the two modes available. 
And yes, if you switch off that flag you'll get the `impedance` mode disabled.

kt10aan, 2015-07-14:
@pattacini thanks!

pattacini, 2015-07-14:
Closing...
For any other doubt, please comment on https://github.com/robotology/icub-basic-demos/issues.

closed 2020-12-15 by pattacini