robotology/community #46: [discussion] Options on how to reduce a YARP port output frequency

plinioMoreno, 2015-07-15:
Given a YARP module that outputs some data at a given frequency (that I do not want to reduce), I want to read/visualize some of the outputs (every once in a while, but not all of them). The use case is our mobile robot Vizzy that is connected through wireless and we want to see the images every second to avoid network saturation. This said, we do not want to read the actual image port, but a port that somehow reduces the output frequency of the image port.

Are there libraries/modules for that in YARP? Otherwise how would be the best way to do this?

pattacini, 2015-07-15:
Hi @plinioMoreno 

See if [`yarp sample`](http://wiki.icub.org/yarpdoc/yarp.html#yarp_sample) can be used in your case.

plinioMoreno, 2015-07-15:
Thanks @pattacini , yarp sample does the work properly.

drdanz, 2015-07-16:
`yarp sample` creates a process that reads on a port and outputs on another port with a different frequency. This means that you have an extra process, an extra connection (meaning that the data is sent twice, even though one of the connections is likely to be local), and (if `yarp sample` is not running on the same computer as the sender process) extra data on the network.

Another option is to use the [portmonitor](http://wiki.icub.org/yarpdoc/portmonitor.html) to modify the rate directly on the sender connection, see [this example](https://github.com/robotology/yarp/blob/master/example/portmonitor/port_rate/rate.lua). It is written in lua, but you can write a small compiled plugin if you don't want to use lua.

plinioMoreno, 2015-07-17:
Hi @drdanz ,

thanks for the suggestion. I'll give it a try and let you know.

closed 2020-12-08 by pattacini