robotology/community #47: installation on mac - issue

serena-ivaldi, 2015-07-20:
Hi guys, I am re-installing everything from scratch on my Mac, following this guide:
http://wiki.icub.org/wiki/Homebrew_Mavericks

I am finding an error in this formula:
brew install `brew deps --1 yarp | sed -e ':a' -e 'N' -e '$!ba' -e 's/\n/ /g’`

It says:
ivaldi-perso:~ arwen$ brew install `brew deps --1 yarp | sed -e ':a' -e 'N' -e '$!ba' -e 's/\n/ /g'`
Error: No available formula for yarp 
This command requires a formula argument

What should I do?

kt10aan, 2015-07-20:
Hi Serena,

in your case, the issue is that the yarp formula is in the homebrew-x11 repository. So you should do first:
''brew tap homebrew/x11''

However, since the yarp formula was significantly updated recently, I recommend you to go to:
http://wiki.icub.org/yarpdoc/install_yarp_mac.html

and check the instructions there, since a lot of things have changed.

kt10aan, 2015-07-20:
By the way, a short summary of things that have changed:
- X11 is no longer needed
- The yarp formula has options for the GUIs, some yarpdev devices and some bindings, so depending on your needs, you may not have to install from source.
- Also, the dependencies have also changed since the last version of yarp (mostly for the GUIs)

serena-ivaldi, 2015-07-20:
ok thanks! 
however I already have X11 on my mac.. it's weird it was not detected.

please update the link on the wiki as it is the link that people coming to the summerschool are seeing right now!

kt10aan, 2015-07-20:
X11 has nothing to do with it. X11 is not needed any more.
The homebrew/x11 is just a repository (which you add by ``brew tap homebrew/x11``). YARP in not in the main homebrew repository.

kt10aan, 2015-07-20:
Updated the wiki page at: http://wiki.icub.org/wiki/MacOSX:_installation

serena-ivaldi, 2015-07-20:
Thanks!

I have another issue. I am following the update guidelines, and after brewing everything, I am now compiling yarp and got this error in make:

[ 88%] Building CXX object src/yarpdataplayer-gtk/CMakeFiles/yarpdataplayer-gtk.dir/idl_generated_code/src/dataSetPlayer_IDL.cpp.o
make[2]: *** No rule to make target `/usr/local/lib/libpangoxft-1.0.dylib', needed by `bin/yarpdataplayer-gtk'.  Stop.
make[1]: *** [src/yarpdataplayer-gtk/CMakeFiles/yarpdataplayer-gtk.dir/all] Error 2
make: *** [all] Error 2

Apparently there is a problem with pango and/or gtk, but it was installed following the instructions. Any help?

kt10aan, 2015-07-20:
I am assuming you are installing YARP from source.
Why are you building the GTK version of yarpdataplayer? Use the Qt5 version instead. Have you installed Qt5? And have you setup the Qt5_DIR environmental variable?
It's all in the instructions (Section:Installing YARP dependencies)

serena-ivaldi, 2015-07-20:
yes I did that ! I don't know why it's asking for GTK..

serena-ivaldi, 2015-07-20:
as a quick fix, i toggled in ccmake to have all the options, and i switched to OFF all the variables of type YARP_USE_GTK* 
By doing this i can compile correctly.. however I don't know if this is the correct way of doing it ;-)

serena-ivaldi, 2015-07-20:
by the way, same problem installing icub-main..
I had to disable ICUB_USE_GTK otherwise plenty of errors..

kt10aan, 2015-07-20:
GTK GUIs are deprecated anyway (except, for now, YarpBuilder). So i do not think that they will keep up with the updated versions of GTK for the Mac.

serena-ivaldi, 2015-07-20:
there is still demoForceControl (which is very useful!)

kt10aan, 2015-07-20:
Well, strictly speaking, demoForceControl is not in the icub-main anymore.
It would be useful though to change the demoForceControl GUI into qt5.

pattacini, 2015-07-25:
@serena-ivaldi can we close it?

closed 2020-12-15 by pattacini