robotology/community #48: How to create a ROS topic in a RateThread, which is inside of an RFModule

plinioMoreno, 2015-07-28:
I want to create a ROS node that has several topics attached to them, with the following structure:

* Each of the subscribers are defined inside a RateThread class, so I have as many threads as subscribers.
* The publishers and the threads are defined inside an RFModule, which is initialized in the main function.

How should I do this? So far I've been able to create a publisher that reads properly the output, but the node information is not available from the roscore, which means that `rostopic list` just shows the topic without the node name.

The code that I've been testing as follows:

```cpp
#include <iostream>
#include <yarp/os/all.h>
#include "actionlib_msgs_GoalID.h"

using namespace std;
using namespace yarp::os;

Mutex key;

class Goal_ID_Thread : public yarp::os::RateThread {
public:
    Goal_ID_Thread(int id):RateThread(id) {
      my_id = id;
      name_defined = false;
    }
    void set_port_name(std::string port_name) {
      goal_port_name = port_name;
      name_defined = true;
    }

    bool threadInit() {
        if (name_defined) {
            goal_id_port.setReadOnly();
            bool outputOk_1 = goal_id_port.topic(goal_port_name);
            if (outputOk_1)
                cout<<"Goal id port initialized properly."<<endl;
            return true;
        }
        else return false;
    }

    void run() {
        if (actionlib_msgs_GoalID *cancel=goal_id_port.read(false)) {
            std::cout << cancel->id << " sec: " <<cancel->stamp.sec << " nsec: " << cancel->stamp.nsec << std::endl;
        }
    }
    void threadRelease() {
        goal_id_port.interrupt();

    }

private:
    int my_id;
    std::string goal_port_name;
    bool name_defined;
    yarp::os::Subscriber<actionlib_msgs_GoalID> goal_id_port;
};

class ActionServerModule:public RFModule
{
    int count;
   Goal_ID_Thread* worker1;

public:

    double getPeriod() {
        return 1; //module periodicity (seconds)
    }

    /*
    * This is our main function. Will be called periodically every getPeriod() seconds.
    */
    bool updateModule()
    {
        return true;
    }

    /*
    * Configure function. Receive a previously initialized
    * resource finder object. Use it to configure your module.
    * Open port and attach it to message handler.
    */
    bool configure(yarp::os::ResourceFinder &rf)
    {
        worker1 = new Goal_ID_Thread(1);
        worker1->set_port_name(std::string("/cancel"));
        cout<<"Staring worker thread..."<<endl;
        worker1->start();
        return true;
    }

    /*
    * Interrupt function.
    */
    bool interruptModule()
    {
        cout<<"Interrupting your module to stop the worker thread..."<<endl;
        worker1->stop();
        return true;
    }

    /*
    * Close function, to perform cleanup.
    */
    bool close()
    {
        cout<<"Calling close function\n";
        delete worker1;
        return true;
    }
};

int main(int argc, char * argv[])
{
    Network yarp;
    yarp::os::Node *action_node;
    action_node = new yarp::os::Node(std::string("/fibonacci"));
    ActionServerModule module;
    ResourceFinder rf;
    rf.configure(argc, argv);
    // rf.setVerbose(true);

    cout<<"Configure module..."<<endl;
    module.configure(rf);
    cout<<"Start module..."<<endl;
    module.runModule();

    cout<<"Main returning..."<<endl;
    return 0;
}
```

The message file was generated by `yarpidl_rosmsg actionlib_msgs/GoalID`

I know that works if I define the subscriber in the main module, but I would like to have the subscriber in the Thread.

plinioMoreno, 2015-07-28:
@barbalberto let me rephrase my question, because it is based on the usual practices when using ROS NodeHandle:

I would like to have a similar behaviour between the `yarp::os::Node` and the `ros::NodeHandle` as follows:

* The `ros::NodeHandle` can be initialized as "namespaced". This means that if I create a topic attached to the handle, the topic name is built as "namespace + /+ topic_name". This is the suggested way of creating [topics in nodes] (https://github.com/ethz-asl/ros_best_practices/wiki#node-handles).

From what I read and the behaviour during execution, it looks like `yarp::os::Node` does not have the "namespaced" option. The only drawback of this is that I cannot create at the same time a `yarp::os::Node` named as `/my_node` and a `yarp:os::Subscriber` named as `/my_node/my_topic`. It would be nice if the case above would be allowed, because this is the usual way of initializing [actionlib servers] (http://wiki.ros.org/actionlib_tutorials/Tutorials/SimpleActionServer%28ExecuteCallbackMethod%29)

My mistake was to associate the node name to the namespace, which is not true. How difficult would be to add the namespace feature to the `yarp::os::Node`?

barbalberto, 2015-07-29:
I'm not sure to understand your point, I'll do my best to answer. I'll add some explanation in the hope this'll be useful also for other people reading.

* You can create topics either in the RFModule or in a thread as long as they are created after the node. All topics will be associated to the node.

In you example the `yarp name list` commands returns:
``` registration name /cancel-@/fibonacci ip 10.255.72.134 port 10010 type tcp ```

meaning the topic is associated with the node, and `rostopic info /fibonacci` returns:
```
Type: actionlib_msgs/GoalID

Publishers: None

Subscribers: 
 * /fibonacci (http://10.255.72.134:10009)
``` 
It looks to me that also ROS understand the association between node and topic. Probably I'm not understanding what you are trying to achieve.

* rostopic list just shows the topic without the node name.
Yes, afaik this is the correct behaviour. From http://wiki.ros.org/it/ROS/Tutorials/UnderstandingTopics the topic namme is not associated with the publisher nor with the subscriber but it is a third separated entity with its own name. If both the publisher and the subscriber will prepend the node name, they'll not be able to talk each other.

* Regarding namespace, if I understood correctly it is a 'prefix' to append to the topic name. Will it be the same for all topics opened by the current node?
How is it usually set in ROS code? I guess it is shared between different modules, is it an environment variable or hard coded?

barbalberto, 2015-07-29:
> The only drawback of this is that I cannot create at the same time a yarp::os::Node named as /my_node and a yarp:os::Subscriber named as /my_node/my_topic

This can be done by explicitly opening the topic with the full name like:
``` subscriber.topic("/my_node/my_topic") ```

You can achive the functionality of adding a prefix to everything both node name and topic name using the YARP_PORT_PREFIX environment variable, for example by running the executable with
```YARP_PORT_PREFIX=/namespace  my_exec```

 ```yarp name list``` will output
```registration name /namespace/cancel-@/namespace/fibonacci ip 10.255.72.134 port 10005 type tcp
registration name /namespace/fibonacci ip 10.255.72.134 port 10004 type tcp```

Then when you do rosnode list and rostopic list
```
$ rostopic list 
/namespace/cancel
/rosout
/rosout_agg
```
```
$ rosnode list 
/namespace/fibonacci
/rosout
```
Is this what meant, or you meant to add the namespace only to topics or only to nodes?
A part from the YARP_PORT_PREFIX, or 'manually' adding the namespace as a prefix while creating the node/topics, I can think of a way to configure the node in such a way it will to automatically add the namespace prefix to its own topics, if you think this will be helpful.

Please let me know if I understood your point.

plinioMoreno, 2015-07-30:
@barbalberto,

The YARP_PORT_PREFIX is what I was looking for! It works exactly as the namespace, and what is great is that works on runtime. The first part of my question was not very clear, but in the second part is easier to understand.

It is the first time I hear about the YARP_PORT_PREFIX. Do you know if there is a pointer in the YARP documentation? I could not find it anywhere.

Thanks for your answer

barbalberto, 2015-08-03:
Actually there is no a dedicated page, however that's all what it does, adds a prefix to each port  ;)
I'll add it to the list of documentation to update.

closed 2020-12-08 by pattacini