robotology/community #49: Java bindings and getIplImage()

kt10aan, 2015-08-27:
Hi,

I am having some trouble converting a YARP RGB Image to IplImage. In C++, everything works fine, but, unfortunately, I have to use Java. But the getIplImage() function in Java returns SWIGTYPE_p_void (and not an actual pointer). My Java skills are rusty at best, so...

In essence, I am trying to reproduce in Java this:
http://wiki.icub.org/yarpdoc/opencv_2main_8cpp-example.html
especially the line with the function cvCvtColor.
There is also the added constraint that the solution should be reasonably fast (i.e., can achieve 25fps)

And no, not using Java for this is unfortunately not an option, as there are many other dependencies... :(

traversaro, 2015-08-27:
I am probably missing something, but can't you use the `getCPtr` method of the class `SWIGTYPE_p_void` to get the C pointer  casted of the IplImage contained to a java long ?  
Once you get the pointer I guess that the problem is to pass the C pointer to the OpenCV Java API, but that is quite YARP-independent.

pattacini, 2015-09-17:
ping @kt10aan on the @traversaro's asnwer.

kt10aan, 2015-09-17:
Ah yes. While @traversaro's answer was quite enlightening, i still haven't managed to solve the issue. Mostly because of many many years that I didn't have to touch Java!. I do get the C pointer in a java long, but I have no idea what to do with it afterwards. If someone has a specific example/code, I would be grateful, as I am stuck on this (even though it is not a yarp issue, but then, this is the QA repository :) ).

traversaro, 2015-09-17:
Mhh in the official Java OpenCV bindings it does not seem that IplImage is wrapped... which OpenCV bindings are you using?

kt10aan, 2015-09-17:
I am using the javacv bindings (https://github.com/bytedeco/javacv). As I mentioned above, because of a collaboration, I do not choose what to use. I am trying to interface someone else's system to yarp.

kt10aan, 2015-09-18:
Ok, the issue was solved (@traversaro's comment led me to the correct thinking)! Just for reference, the key was to use the getRawData instead of get IplImage.
You can close the issue now :)

closed 2020-12-15 by pattacini