robotology/community #51: iCubOsaka01 robotInterface error messages

kt10aan, 2015-09-25:
We updated our robot (software/firmware) and now by running robotInterface, a multitude of error messages show up.

I know things changed on the icub device side the last months, so:

Is it from old config files? Is it because I compiled an extra device or one less? 

Or is it, more likely, me forgetting something obvious? :D

(By the way, the documentation regarding compilation on the PC104 is outdated, regarding the devices, http://wiki.icub.org/wiki/Compilation_on_the_pc104)

Here is the log: https://gist.github.com/kt10aan/84cc6e28772eacc6eadb

Tobias-Fischer, 2015-09-25:
Silly question .. but is a `yarpserver` running somewhere, and is `pc104` setup correctly to talk to that `yarpserver`? :)

kt10aan, 2015-09-25:
Yes, obviously there is a yarp server running.
As for the correctly configured, I double checked by cleaning yarp.conf and then adding the server again in the config.

Funny thing is that it's the first time all these years that I have seen these specific error messages :)

kt10aan, 2015-09-25:
But I do hope of course that it is something so simple as a yarp server issue! :)

Tobias-Fischer, 2015-09-25:
Would have been too easy ;). Sorry, I have no other idea on the spot.

kt10aan, 2015-09-26:
@Tobias-Fischer, sometimes life is good and things are too easy! Somehow there was a lingering yarp server process and a reboot of the icubsrv fixed the issue. Now, if all the problems were like that :)

barbalberto, 2015-09-28:
By the way, for the warning message:
************************************************************************************************
  AnalogServer is using deprecated parameters for port name! It should be:
       name:         full name of the port, like /robotName/deviceId/sensorType:o
       period:       refresh period of the broadcasted values in ms (optional, default 20ms)
************************************************************************************************ 

it is better to update the config file following the message instructions, which means
the 'name' parameter in the analogserver config file should contains the full name of the yarp port and 'period' parameter needs to be the desired broadcast period in ms.

kt10aan, 2015-09-28:
@barbalberto, thanks! I will update the config files.

closed 2020-12-15 by pattacini