robotology/community #53: Encoders and MotorEncoders readings

hu-yue, 2015-09-29:
Hi,
I used iEncoders and iMotorEncoders to read the encoders of the robot, as I wanted to test the effect of the springs in the joints.
However, I'm not sure about how to interpret the readings from iMotorEncoders. I thought it is enough to divide by the gear ration (100), but the numbers I obtained are not the same as iEncoders even when the springs are not mounted.
And as I've understood, iEncoders and iMotorEncoders should read the same values (almost?) when no spring is mounted.

What is the correct way to interpret the readings of iMotorEncoders?
@DanielePucci was also interested in this.

Thanks!

DanielePucci, 2015-09-29:
@randaz81

francesco-romano, 2015-09-29:
Have you tried to see what  [getGearboxRation](http://wiki.icub.org/yarpdoc/classyarp_1_1dev_1_1IMotor.html#a6abbd0ad79a61b17872f4c0f513654e7) returns?

hu-yue, 2015-09-29:
Nope, can try that.

randaz81, 2015-09-29:
Yes, indeed you have to divide the position returned by iMotorEncoders by the gear ratio obtained by getGearBoxRatio, considering also the sign. Additionally, you have to remove the offset which affects the motor encoder (indeed they are relative encoder, not absolute encoders). check `opticalEncoders-consistency` in `icub-tests` for an example

hu-yue, 2015-09-29:
I get false from getGearboxRatio and so nonsense gear ratio, is the integer number m of the arguments ranging from 0 to 5 for example in the case of left_leg? 
```
getGearboxRatio (int m, double *val)
```

randaz81, 2015-09-29:
yes, it is. For the legs legs it should return *val= +-100

hu-yue, 2015-09-29:
Ok, but it return false and val = 0. And also the function ```getNumberOfMotors (int *num)``` returns false and num is 0. 
But this part did not give any error:
```
    Property options;
    options.put("device", "remote_controlboard");
    options.put("local", localPorts.c_str());
    options.put("remote", remotePorts.c_str());
    PolyDriver robotDevice(options);
    IMotor *motors;
    if (!robotDevice.view(motors)) {
        printf("Problems acquiring interfaces\n");
        return 0;
    }
```
And the iEncoders and iMotorEncoders seem to work fine.

randaz81, 2015-09-29:
Looking at git history, icub-main has be updated at least at 04 July to properly manage getGearboxRatio. Is it so?

hu-yue, 2015-09-29:
Ok, we'll talk about the update after IROS and I'll try then. Thanks!

hu-yue, 2015-09-29:
icub-main is updated up to 04/09/2015, as well as yarp. But the firmware is from June.

pattacini, 2015-10-07:
@hu-yue any progress on this?

hu-yue, 2015-10-07:
@pattacini not yet, sorry, I just got back from Hamburg but have to leave for Delft already so I did not have time to check the firmware issue with @randaz81, maybe we can do it next week?

pattacini, 2015-10-07:
there's no hurry, was just a ping :smirk:

hu-yue, 2015-10-15:
After the last firmware and software update now it's possible to read the gear ratio using ```getGearboxRatio (int m, double *val)```.
The motor encoders values were converted into encoders values as
```(m_encoders - enc_offsets)/gear_ratio```

randaz81, 2015-10-22:
yes 10^-2 deg is ok. 
btw, the module `opticalEncoders-consistency` has been recently renamed to `motorEncoders-consistency`. This test is more complete in the sense that checks also for the consistency between the joint velocity and the motor velocity.

hu-yue, 2015-10-22:
Ok thanks!

traversaro, 2016-03-31:
fyi @fjandrad

closed 2020-12-15 by pattacini