robotology/community #55: CAD drawings for the F/T sensors

iron76, 2015-10-07:
I am trying to find the CAD drawings for the F/T sensors described here (http://wiki.icub.org/wiki/FT_sensor). Can anyone give me a pointer to repository were these drawings can be downloaded?
Thanks in advance.

traversaro, 2015-10-07:
I think it should be this one: https://svn.robotology.eu/repos/iCubHardware-pub/trunk/mechanics/public/icub1_icub2/ftsensor45a.asm.1 .

iron76, 2015-10-07:
What if I only need to share with someone else only the dimensions and the screw positions for designing a socket for the FT sensor? Is there an simpler format to share?

traversaro, 2015-10-07:
Then I think I am unable to reply, and we need some help from some mechanics expert (@florisi ? ).

fiorisi, 2015-10-07:
You can find the dxf files for the ftsensor45 in this folder:
https://svn.robotology.eu/repos/iCubHardware/trunk/mechanics/public/icub1_icub2/prodoc

where:
- ftsensor45a.dxf                  --> assembly 
- ftsensor45.dxf                    --> sensor cell
- bottomcover45_r_A1.dxf   --> bottom cover
- topcover45.dxf                   --> top cover

iron76, 2015-10-08:
@fiorisi, can we also have the details of the screws you are using to mount the FT sensor. Is the diameter M2.5? Thanks in advance.

fiorisi, 2015-10-09:
Both the top and bottom cover have M3 holes.

fiorisi, 2015-10-09:
[image]

pattacini, 2015-10-14:
@iron76, is @fiorisi 's answer satisfactory?

iron76, 2015-10-14:
Yes, I will close the issue.

closed 2020-12-15 by pattacini