robotology/community #56: Weight of iCubGenova02 (Purpley)

DanielePucci, 2015-10-07:
Does anyone know the weight of iCubGenova02 (Purpley) with and without the battery? In the case of no battery, it would be nice to have the robot's weight with the dummy.

DanielePucci, 2015-10-07:
Yes, I know.

The weight of the robot without the battery and with the dummy is 
~~~
32.7 Kg
~~~
[image]

The weight of the dummy is 
~~~
1.3 Kg
~~~
[image]

while the weight of the battery is 
~~~
1.7 Kg
~~~
[image]

So, the weight of the robot with the battery is:
~~~
33.1 Kg
~~~

DanielePucci, 2015-10-07:
@iron76 @francesco-romano @traversaro @maggia80

traversaro, 2015-10-07:
For the sake of completeness, did you checked the weight of the soles ? I would expect they are under 100 grams and can be assumed as negligible, but it is always nice to check.

DanielePucci, 2015-10-07:
Well, the soles' weight is 
~~~
0.2 Kg
~~~
[image]

So, the robot's weight with the battery and the soles is
~~~
33.3 Kg
~~~

pattacini, 2015-10-07:
As usual, @randaz81 always shows up in the pictures we take :smile:

DanielePucci, 2015-10-07:
Really focused this times :+1:

DanielePucci, 2015-10-09:
Additional pictures on the weight of single components.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

DanielePucci, 2015-10-27:
Shoulder-arm groups:
[image]
[image]

@traversaro @francesco-romano @Mick3Lozzo

DanielePucci, 2015-11-19:
@spada1978

spada1978, 2015-11-19:
Lowerbody 
weight 15.15 Kg
[image]
[image]
[image]

spada1978, 2015-11-19:
Lowerbody mechanics / electronics
weight 16.44 Kg
[image]
 
[image]
[image]

randaz81, 2015-12-04:
The dummy weight has been updated and now is identical to the battery weight (1.7Kg). Now installed on iCubGenova02.

closed 2020-12-15 by pattacini