robotology/community #57: Sensing current in iCub motors

jypuigbo, 2015-10-07:
Do you think it is possible to stream in a yarp port the output current (in Amperes) of the different motors, in order to get some feedback of physical obstacles when moving?

If that is the case, where is it available or what should I do to make it available? 

Thank you, 

Jordi

barbalberto, 2015-10-07:
Right now the currents are not streamed from the stateExt port but you can read the joint torques (the same you can see from the motorGui) which is not the same but is somehow related. I don't know if it is enough in your case.

In order to have currents you have to request them using rpc port (which is kind of slow so it is not suggested to do it at high rate). You can use the controlBoardDumper with the command:

controlBoardDumper  --robot icub --part left_arm --rate 10  --joints "(0 1 ... n)" --dataToDump "(getCurrents)"

It'll create a port streaming the current values you can read from. Since a man-in-the-middle is added, the delay will increase a bit.

jypuigbo, 2015-10-07:
Thanks a lot, @barbalberto . I think this will do, at least by now.

iron76, 2015-10-08:
Dear Jordi, if you have an iCub version 1.x (the one with CAN-bus control boards) you can use the controBoardDumper (http://wiki.icub.org/brain/group__icub__controlBoardDumper.html). It requires enabling the broadcast of currents: http://wiki.icub.org/wiki/ControlBoard_configuration_file#Broadcast

closed 2020-12-15 by pattacini