robotology/community #60: ESD usb-canbus

iron76, 2015-10-09:
I am trying to use the `icubmod_ecan` module on Ubuntu 14.04. I have a couple of questions that I couldn't find in the available documentation.

1. Which versions of the ESD CAN-USB device are compatible with my Ubuntu release? In other words, which ESD CAN-USB hardware is provided with a driver for my Ubuntu release?

2. Where can I find the drivers for the ESD CAN-USB devices? I found this page http://eris.liralab.it/wiki/Esd_Can_Bus but it seems quite old and it might not be updated.

Thanks in advance.

lornat75, 2015-10-09:
I am not sure, but I think the driver to use depends on the version of the USB dongle you have. The old "gray" mini USB dongle uses the ecan and the driver you found, the newer device (blue)  uses the socket can device driver and the corresponding YARP driver:

 https://github.com/robotology/icub-main/tree/master/src/libraries/icubmod/socketCan

cc @maggia80 @randaz81

randaz81, 2015-10-09:
To my knowledge only the new ESDCAN2 https://esd.eu/en/products/can-usb2 is supported by Ubuntu. You need to use the `socketcan` yarp wrapper driver. You don't kneed a specific linux driver, it's built-in in the kernel. Ssome instructions here: http://wiki.icub.org/wiki/IKart#Esdcan_driver

closed 2020-12-15 by pattacini