robotology/community #61: Wight of iCub(s)

DanielePucci, 2015-10-11:
| iCubGenova01 (v. 2.0) | iCubGenova02 (v. 2.5) with battery | iCubGenova02 (v. 2.5) with dummy | iCubGenova03 (v. 1.0)|
|--------------------|-------------------|-------------------|-------------------|
|             30.3     |             33.1      |             32.6      |             27.5      |

The iCubs were weighted without the skin under the feet's soles. In the case it is installed, add to the above weights `0.2 Kg` (see #56)

@traversaro @francesco-romano @maggia80 @fabrizio69 @super-ste (please inform Michele Gesino)

pattacini, 2015-10-11:
Finally I know what exact numbers to give when guests will asking for those weights!

Thanks @DanielePucci

DanielePucci, 2015-10-11:
:dancers:

super-ste, 2015-10-12:
Please remove version tag (except from iCubGenova02) they're misleading.
The only way to classify our robot is givin' out the exact description of the way they're made.
e.g. for iCubGenova01
iCub V1 Body 
head V2
SEA legs (without SEA)
:sweat_smile:

DanielePucci, 2015-10-12:
Ack.

alecive, 2015-10-12:
@DanielePucci it would be nice if you modify your first post according to @super-ste 's suggestions :+1:

DanielePucci, 2015-10-12:
Done, checks are welcome.

dogoepp, 2015-10-23:
Hi Daniele,

What about putting all this weight information in the official 
documentation of iCub ? I think that it might be easier to find there.

DanielePucci, 2015-10-27:
Hi @dogoepp,
nice idea. Volunteers are welcome.

DanielePucci, 2015-10-27:
[image]
[image]

traversaro, 2015-10-27:
For the weights on `iCubGenova01` and `iCubGenova03` the photos were shot on 30/04/2014, so they reflect the situation of the robots at that point in time.

DanielePucci, 2015-11-20:
For consistency checks, it would be awesome if our friends in Paris @rlober @mingxing-liu could weight their robots :)

francesco-romano, 2015-11-21:
Added iCubDarmstadt01 as in https://github.com/robotology/icub-support/issues/118

DanielePucci, 2015-11-21:
:100:

rlober, 2015-12-04:
@DanielePucci 

Next time I am down in the cave I will check our iCubs' body mass indexes!

DanielePucci, 2016-03-09:
@spada1978

DanielePucci, 2016-03-09:
The dummy has been re-designed. Then, the weight is changed accordingly.

DanielePucci, 2016-03-09:
For further information on the previous dummy weight, see #56

traversaro, 2016-03-09:
@DanielePucci what is the status of the models commited in the repositories?

DanielePucci, 2016-03-09:
I sincerely don't remember, but I have flashes of a conversation with you where I asked you to add additional weight on the dummy. We shall check @traversaro @francesco-romano

gabrielenava, 2017-05-31:
Last week I weighed the green iCub (iCubGenova04, v2.5 with battery pack) without the skin under the feet's soles and this is the result:

[image]

i.e., the same the weight of the purple (as expected)

fjandrad, 2018-09-04:
[image]

iCubGenova04 on the scale. 
**Total weight = 33.33kg**
@DanielePucci

DanielePucci, 2018-09-04:
Nice @fjandrad. Can we however update the table on the first https://github.com/robotology/QA/issues/61#issue-110842856?

closed 2020-12-15 by pattacini