robotology/community #62: What does yarp::dev::IVelocityControl::stop() actually do at the lowest level?

pattacini, 2015-10-13:
We all know that velocity commands are integrated by the `FW` to retrieve position references to follow by means of the `PID`.

Whenever a `yarp::dev::IVelocityControl::stop()` is issued, what actually happens at the lowest level? Two possibilities are given:

1. The FW simply stops integrating as it had received a `velocityMove(0.0)`.
2. The FW sets the internal position reference equal to the current encoder reading.

Is there any third one?

I'm a bit concerned about hypothesis **1**: because of a poorly tuned PID, the joint will be still moving for a while after the `stop` command is received.

I hope the implementation resorts to **2**, indeed.

##### Second question
Does the _joints subset_ version of the `stop` command travel over the _rpc_ channel or over the _streaming_ channel? Can we be sure that the `stop` command is never lost?

/cc @randaz81 @barbalberto

pattacini, 2015-10-13:
##### Backstage 
I'm asking this because while teleoperating the iCub, I've realized that the fingers - which are controlled in velocity and whose control is a bit degraded - do not stop exactly upon `stop` reception but, instead, proceed moving for a short while afterwards.

ale-git, 2015-10-13:
I can answer about the joints controlled by 2FOC boards (thus not regarding the fingers), where the stopping method is 2 (new reference = encoder reading).

pattacini, 2015-10-13:
@ale-git cool thanks :+1: 
What about the `MC4` (controlling the fingers) then?

barbalberto, 2015-10-13:
All ```stop``` command are send using ```rpc```.

pattacini, 2015-10-13:
Thanks @barbalberto :+1: 
Thus, only one doubt (the main one) remaining.

randaz81, 2015-10-13:
The implementation 2 is already used on MC4 boards. 
```
	if (_control_mode[axis] == MODE_IMPEDANCE_POS || \
		_control_mode[axis] == MODE_POSITION || \
		_control_mode[axis] == MODE_IMPEDANCE_VEL || \
		_control_mode[axis] == MODE_VELOCITY || \
		_control_mode[axis] == MODE_MIXED) \
		{ \
		    _set_vel[axis] = 0; \
		    _set_acc[axis] = 0; \
            _desired_vel[axis] = 0; \
			init_trajectory (axis, _position[axis], _position[axis], 1); \
		} \
```

pattacini, 2015-10-13:
Great, thanks @randaz81 
Hence I'd need to better inquiry what I've been observing with the fingers!

closed 2020-12-08 by pattacini