robotology/community #63: Updating Yarp

jpiabrantes, 2015-10-16:
I had my Yarp working but I decided to update it by doing `git pull`, I did `make clean`, `cmake..` and ran `make` again. I got a very strange error on the `make`:

```
[ 97%] Building CXX object src/yarpdataplayer-gtk/CMakeFiles/yarpdataplayer-gtk.dir/src/worker.cpp.o
/home/joao/github/yarp/src/yarpdataplayer-gtk/src/worker.cpp: In member function ‘int WorkerClass::sendImages(int, int)’:
/home/joao/github/yarp/src/yarpdataplayer-gtk/src/worker.cpp:132:5: error: ‘string’ was not declared in this scope
     string tmpPath = utilities->partDetails[part].path;
     ^
/home/joao/github/yarp/src/yarpdataplayer-gtk/src/worker.cpp:132:5: note: suggested alternative:
In file included from /usr/include/c++/4.8/string:39:0,
                 from /home/joao/github/yarp/src/libYARP_OS/include/yarp/os/ConstString.h:32,
                 from /home/joao/github/yarp/src/libYARP_OS/include/yarp/os/PortReader.h:13,
                 from /home/joao/github/yarp/src/libYARP_OS/include/yarp/os/Portable.h:13,
                 from /home/joao/github/yarp/src/libYARP_OS/include/yarp/os/Stamp.h:15,
                 from /home/joao/github/yarp/src/yarpdataplayer-gtk/src/worker.cpp:24:
/usr/include/c++/4.8/bits/stringfwd.h:62:33: note:   ‘std::string’
   typedef basic_string<char>    string;   
                                 ^
/home/joao/github/yarp/src/yarpdataplayer-gtk/src/worker.cpp:132:12: error: expected ‘;’ before ‘tmpPath’
     string tmpPath = utilities->partDetails[part].path;
            ^
/home/joao/github/yarp/src/yarpdataplayer-gtk/src/worker.cpp:133:12: error: expected ‘;’ before ‘tmpName’
     string tmpName;
            ^
/home/joao/github/yarp/src/yarpdataplayer-gtk/src/worker.cpp:135:9: error: ‘tmpName’ was not declared in this scope
         tmpName = utilities->partDetails[part].bot.get(frame).asList()->tail().tail().get(1).asString().c_str();
         ^
/home/joao/github/yarp/src/yarpdataplayer-gtk/src/worker.cpp:137:9: error: ‘tmpName’ was not declared in this scope
         tmpName = utilities->partDetails[part].bot.get(frame).asList()->tail().tail().get(0).asString().c_str();
         ^
/home/joao/github/yarp/src/yarpdataplayer-gtk/src/worker.cpp:139:5: error: ‘tmpPath’ was not declared in this scope
     tmpPath = tmpPath + tmpName;
     ^
/home/joao/github/yarp/src/yarpdataplayer-gtk/src/worker.cpp:139:25: error: ‘tmpName’ was not declared in this scope
     tmpPath = tmpPath + tmpName;
                         ^
/home/joao/github/yarp/src/yarpdataplayer-gtk/src/worker.cpp:153:58: error: ‘cvCopyImage’ was not declared in this scope
         cvCopyImage( img, (IplImage *) temp.getIplImage());
                                                          ^
make[2]: *** [src/yarpdataplayer-gtk/CMakeFiles/yarpdataplayer-gtk.dir/src/worker.cpp.o] Error 1
make[1]: *** [src/yarpdataplayer-gtk/CMakeFiles/yarpdataplayer-gtk.dir/all] Error 2
make: *** [all] Error 2
```

Can anybody help me?
I am using Ubuntu 14.04

drdanz, 2015-10-16:
Is opencv installed on your system?
Can you please run "`VERBOSE=1 make`" and report the command line that is been used to build this file?

jpiabrantes, 2015-10-16:
OpenCV is isntalled.
@drdanz is this what you want?:
```
[ 97%] Building CXX object src/yarpdataplayer-gtk/CMakeFiles/yarpdataplayer-gtk.dir/src/worker.cpp.o
cd /home/joao/github/yarp/build/src/yarpdataplayer-gtk && /usr/bin/c++   -DHAS_OPENCV -DYARP_PRESENT -D_REENTRANT -Wall -Wextra -Wsign-compare -Wpointer-arith -Winit-self -Wnon-virtual-dtor -Wcast-align -Wvla  -Wno-unused-parameter -Wno-long-long -Wno-cast-align -Wdeprecated-declarations -O3 -DNDEBUG -fPIE -I/home/joao/github/yarp/src/yarpdataplayer-gtk/include -I/home/joao/github/yarp/build/generated_include -I/home/joao/github/yarp/src/libYARP_OS/include -I/home/joao/github/yarp/src/libYARP_sig/include -I/usr/include/gtk-2.0 -I/usr/include/freetype2 -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/atk-1.0 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/cairo -I/usr/include/pango-1.0 -I/usr/lib/x86_64-linux-gnu/gtk-2.0/include -I/usr/include/sigc++-2.0 -I/usr/lib/x86_64-linux-gnu/sigc++-2.0/include -I/usr/include/glibmm-2.4 -I/usr/lib/x86_64-linux-gnu/glibmm-2.4/include -I/usr/include/giomm-2.4 -I/usr/lib/x86_64-linux-gnu/giomm-2.4/include -I/usr/include/atkmm-1.6 -I/usr/include/cairomm-1.0 -I/usr/lib/x86_64-linux-gnu/cairomm-1.0/include -I/usr/include/pangomm-1.4 -I/usr/lib/x86_64-linux-gnu/pangomm-1.4/include -I/usr/include/gdkmm-2.4 -I/usr/lib/x86_64-linux-gnu/gdkmm-2.4/include -I/usr/include/gtkmm-2.4 -I/usr/lib/x86_64-linux-gnu/gtkmm-2.4/include -I/usr/local/include/opencv -I/usr/local/include -I/home/joao/github/yarp/src/yarpdataplayer-gtk/idl_generated_code -I/home/joao/github/yarp/src/yarpdataplayer-gtk/idl_generated_code/include    -o CMakeFiles/yarpdataplayer-gtk.dir/src/worker.cpp.o -c /home/joao/github/yarp/src/yarpdataplayer-gtk/src/worker.cpp
/home/joao/github/yarp/src/yarpdataplayer-gtk/src/worker.cpp: In member function ‘int WorkerClass::sendImages(int, int)’:
/home/joao/github/yarp/src/yarpdataplayer-gtk/src/worker.cpp:132:5: error: ‘string’ was not declared in this scope
     string tmpPath = utilities->partDetails[part].path;
     ^
```

vtikha, 2015-10-22:
@jpiabrantes I have just committed some changes to the gtk version of the yarpdataplayer. Could you please try it again? thanks

vtikha, 2015-10-22:
@jpiabrantes ok, perfect. Sorry you had to reinstall everything.

vtikha, 2015-10-22:
Closing this issue

jpiabrantes, 2015-11-02:
@vtikha I tried to recompile `yarp` again and this issue reappeared. I must have also re-installed the software that was conflicting with `yarp`.

vtikha, 2015-11-02:
Hi @jpiabrantes you confirm that the error is still the following?

```
[ 97%] Building CXX object src/yarpdataplayer-gtk/CMakeFiles/yarpdataplayer-gtk.dir/src/worker.cpp.o
/home/joao/github/yarp/src/yarpdataplayer-gtk/src/worker.cpp: In member function ‘int WorkerClass::sendImages(int, int)’:
/home/joao/github/yarp/src/yarpdataplayer-gtk/src/worker.cpp:132:5: error: ‘string’ was not declared in this scope
     string tmpPath = utilities->partDetails[part].path;
     ^
/home/joao/github/yarp/src/yarpdataplayer-gtk/src/worker.cpp:132:5: note: suggested alternative:
In file included from /usr/include/c++/4.8/string:39:0,
                 from /home/joao/github/yarp/src/libYARP_OS/include/yarp/os/ConstString.h:32,
                 from /home/joao/github/yarp/src/libYARP_OS/include/yarp/os/PortReader.h:13,
                 from /home/joao/github/yarp/src/libYARP_OS/include/yarp/os/Portable.h:13,
                 from /home/joao/github/yarp/src/libYARP_OS/include/yarp/os/Stamp.h:15,
                 from /home/joao/github/yarp/src/yarpdataplayer-gtk/src/worker.cpp:24:
/usr/include/c++/4.8/bits/stringfwd.h:62:33: note:   ‘std::string’
   typedef basic_string<char>    string;   
                                 ^
/home/joao/github/yarp/src/yarpdataplayer-gtk/src/worker.cpp:132:12: error: expected ‘;’ before ‘tmpPath’
     string tmpPath = utilities->partDetails[part].path;
            ^
/home/joao/github/yarp/src/yarpdataplayer-gtk/src/worker.cpp:133:12: error: expected ‘;’ before ‘tmpName’
     string tmpName;
            ^
/home/joao/github/yarp/src/yarpdataplayer-gtk/src/worker.cpp:135:9: error: ‘tmpName’ was not declared in this scope
         tmpName = utilities->partDetails[part].bot.get(frame).asList()->tail().tail().get(1).asString().c_str();
         ^
/home/joao/github/yarp/src/yarpdataplayer-gtk/src/worker.cpp:137:9: error: ‘tmpName’ was not declared in this scope
         tmpName = utilities->partDetails[part].bot.get(frame).asList()->tail().tail().get(0).asString().c_str();
         ^
/home/joao/github/yarp/src/yarpdataplayer-gtk/src/worker.cpp:139:5: error: ‘tmpPath’ was not declared in this scope
     tmpPath = tmpPath + tmpName;
     ^
/home/joao/github/yarp/src/yarpdataplayer-gtk/src/worker.cpp:139:25: error: ‘tmpName’ was not declared in this scope
     tmpPath = tmpPath + tmpName;
                         ^
/home/joao/github/yarp/src/yarpdataplayer-gtk/src/worker.cpp:153:58: error: ‘cvCopyImage’ was not declared in this scope
         cvCopyImage( img, (IplImage *) temp.getIplImage());
                                                          ^
make[2]: *** [src/yarpdataplayer-gtk/CMakeFiles/yarpdataplayer-gtk.dir/src/worker.cpp.o] Error 1
make[1]: *** [src/yarpdataplayer-gtk/CMakeFiles/yarpdataplayer-gtk.dir/all] Error 2
make: *** [all] Error 2
```

I will try it out now on a similar system as yours.
Vadim

jpiabrantes, 2015-11-02:
@vtikha yes it's exactly the same error.

vtikha, 2015-11-02:
@jpiabrantes could you confirm that you have updated yarp from the git repository recently? As this [commit] (https://github.com/robotology/yarp/commit/f684bcbb86516c507e1d2ffe52857215b346fd59) should have fixed this.

jpiabrantes, 2015-11-03:
@vtikha I thought I had made `git pull` and updated it but I did it again just to be sure and now the error is indeed different (I ran make with `VERBOSE=1`):

```sh
[ 97%] Building CXX object src/yarpdataplayer-gtk/CMakeFiles/yarpdataplayer-gtk.dir/src/worker.cpp.o
cd /home/joao/github/yarp/build/src/yarpdataplayer-gtk && /usr/bin/c++   -DHAS_OPENCV -DYARP_PRESENT -D_REENTRANT -Wall -Wextra -Wsign-compare -Wpointer-arith -Winit-self -Wnon-virtual-dtor -Wcast-align -Wvla  -Wno-unused-parameter -Wno-long-long -Wno-cast-align -Wdeprecated-declarations -O3 -DNDEBUG -fPIE -I/home/joao/github/yarp/src/yarpdataplayer-gtk/include -I/home/joao/github/yarp/build/generated_include -I/home/joao/github/yarp/src/libYARP_OS/include -I/home/joao/github/yarp/src/libYARP_sig/include -isystem /usr/include/gtk-2.0 -isystem /usr/include/freetype2 -isystem /usr/include/glib-2.0 -isystem /usr/lib/x86_64-linux-gnu/glib-2.0/include -isystem /usr/include/atk-1.0 -isystem /usr/include/gdk-pixbuf-2.0 -isystem /usr/include/cairo -isystem /usr/include/pango-1.0 -isystem /usr/lib/x86_64-linux-gnu/gtk-2.0/include -isystem /usr/include/sigc++-2.0 -isystem /usr/lib/x86_64-linux-gnu/sigc++-2.0/include -isystem /usr/include/glibmm-2.4 -isystem /usr/lib/x86_64-linux-gnu/glibmm-2.4/include -isystem /usr/include/giomm-2.4 -isystem /usr/lib/x86_64-linux-gnu/giomm-2.4/include -isystem /usr/include/atkmm-1.6 -isystem /usr/include/cairomm-1.0 -isystem /usr/lib/x86_64-linux-gnu/cairomm-1.0/include -isystem /usr/include/pangomm-1.4 -isystem /usr/lib/x86_64-linux-gnu/pangomm-1.4/include -isystem /usr/include/gdkmm-2.4 -isystem /usr/lib/x86_64-linux-gnu/gdkmm-2.4/include -isystem /usr/include/gtkmm-2.4 -isystem /usr/lib/x86_64-linux-gnu/gtkmm-2.4/include -I/usr/local/include/opencv -I/usr/local/include -I/home/joao/github/yarp/src/yarpdataplayer-gtk/idl_generated_code -I/home/joao/github/yarp/src/yarpdataplayer-gtk/idl_generated_code/include    -o CMakeFiles/yarpdataplayer-gtk.dir/src/worker.cpp.o -c /home/joao/github/yarp/src/yarpdataplayer-gtk/src/worker.cpp
/home/joao/github/yarp/src/yarpdataplayer-gtk/src/worker.cpp: In member function ‘int WorkerClass::sendImages(int, int)’:
/home/joao/github/yarp/src/yarpdataplayer-gtk/src/worker.cpp:154:58: error: ‘cvCopyImage’ was not declared in this scope
         cvCopyImage( img, (IplImage *) temp.getIplImage());
                                                          ^
make[2]: *** [src/yarpdataplayer-gtk/CMakeFiles/yarpdataplayer-gtk.dir/src/worker.cpp.o] Error 1
make[2]: Leaving directory `/home/joao/github/yarp/build'
make[1]: *** [src/yarpdataplayer-gtk/CMakeFiles/yarpdataplayer-gtk.dir/all] Error 2
make[1]: Leaving directory `/home/joao/github/yarp/build'
make: *** [all] Error 2
```

Should I post this error on another issue?
Thanks for the help.

vtikha, 2015-11-03:
@jpiabrantes Can you please `git pull` again and try out a change I have made? If we still have issues it would be easier if I can remotely connect to your system and understand what is going on.

pattacini, 2015-11-11:
@jpiabrantes any news?

jpiabrantes, 2015-11-16:
All good now. Thanks

pattacini, 2015-11-16:
Great :+1:

closed 2020-12-15 by pattacini