robotology/community #64: How to increase torque saturation threshold

hu-yue, 2015-10-16:
Hi,
I would like to increase the threshold of torque saturation of iCubHeidelberg01. Where exactly should this be done?
There is a "TorqueMax" in the configuration xml files under the FT folder, is this the right parameter to change?
Thanks!

DanielePucci, 2015-10-16:
@randaz81

hu-yue, 2015-10-19:
I tried to increase the TorqueMax and restarted the robot but it seems that nothing changed.

traversaro, 2015-10-19:
I checked the source code (https://github.com/robotology/icub-main/blame/f10921a6889ff19ead17e123672ad6f37892565a/src/libraries/icubmod/embObjMotionControl/embObjMotionControl.cpp#L1383) and apparently (as far I can see) TorqueMax is the right parameter. Perhaps @barbalberto knows more.

DanielePucci, 2015-10-22:
@hu-yue may you try to do the same test with a lower saturation limit?

barbalberto, 2015-10-22:
The TorqueMax related to the reading of values, but is not related to the maximum torque the motor can produce.
Can you post the output of the robotinterface to see the exact error message?

hu-yue, 2015-10-22:
Ok, I tried with the following limits:
- 50, the saturation warning was again on 24. In this case, the warning appeared several times before the joint went to hw fault.
- 24 (default that was in the file), saturation on 24. In this case, the warning appeared and the joint went immediately on hw fault.
- 20, saturation on 20. As the first case, also in this case the warning appeared several times before hw fault.

The exact warning we get is:
```
embObjMotionControl::updateMeasure() torque measure saturated to +20.0000 on joint 1, count: 1198
```
The errors following this are:
```
from BOARD 6, src LOCAL, adr 0, time 284s 770m 577u: (code 0x0200000c, par16 0x0001 par64 0x0000000100020000) -> MC: generic motor error (see 64 bit mask parameter) + INFO = NO MORE
```
```
Unable to set the control mode of board 6 joint 1 in HW_FAULT
```

The experiment is the one described in my previous comment, the joint that goes on hw fault is left hip roll.

ale-git, 2015-10-22:
The reported error is a I2T limit fault, i.e. the motor is absorbing too much current - it is a thermal protection.

hu-yue, 2015-10-22:
Is this a new feature? It never happened before

DanielePucci, 2015-10-22:
So, they are two different problems:
- the saturation cannot be increased above  `24`, but it can be decreased. This is being investigated;
- the joint runs into `HF` because of the - incomprehensible -  error @marcoaccame @ale-git 

```
from BOARD 6, src LOCAL, adr 0, time 284s 770m 577u: (code 0x0200000c, par16 0x0001 par64 0x0000000100020000) -> MC: generic motor error (see 64 bit mask parameter) + INFO = NO MORE
```

As for the first, the warning message is 
```
embObjMotionControl::updateMeasure() torque measure saturated to +20.0000 on joint 1, count: 1198
```
but the `joint 1` may be that of any part, e.g. that of the ankle. @hu-yue, may you increase the limit of all the joints' leg? Also, @barbalberto has just added the info on the part, so if you can update  `icub-main` before testing again it would be great :)

ale-git, 2015-10-22:
No, it is a protection that has always been implemented in the motor board... have you ever tried the same task with the other leg? Maybe some hardware problem in the motor has arised.

DanielePucci, 2015-10-22:
In light of the above,  @hu-yue, can you open another issue on  `icub-support `, e.g.  `Weird hardware fault `?

hu-yue, 2015-10-22:
I never tried with the other leg. I cannot exclude the possibility of motor problems, lately the robot has been used much more than before.

@DanielePucci I will open another issue for the hw fault.

barbalberto, 2015-10-22:
The correct way to change the torque measure saturation threashold is to modify the value 
```TorqueMax``` in the config file under
```robotname/hardware/mechanicals/part*.xml ```

DanielePucci, 2015-11-19:
Are we sure that the parameter `TorqueMax` should be under `mechanicals`? It is not really clear to me why a `torque` is considered as a mechanical part.

iron76, 2015-11-20:
I would say that the maximum torque is related to the overall mechanical structure of the joint group. This includes the motor but also the motor shaft, the bearings and other optional components such as the tendons. If `TorqueMax` refers only to motors than it should go in the `motorSpecs` folder.

francesco-romano, 2015-11-23:
cc @marcoaccame 
What about the EMS number representation?

closed 2020-12-08 by pattacini