robotology/community #66: iCub Python bindings

jpiabrantes, 2015-10-18:
I was having some troubles installing `yarp` so I re-installed Ubuntu and started from a fresh version. I was able to install `yarp` and `icub` from the source files. I installed the python bindings on `yarp` but I wasn't able to install the `icub` python bindings. When I do `import icub` I get:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/joao/github/icub-main/bindings/python/icub.py", line 28, in <module>
    _icub = swig_import_helper()
  File "/home/joao/github/icub-main/bindings/python/icub.py", line 24, in swig_import_helper
    _mod = imp.load_module('_icub', fp, pathname, description)
ImportError: /home/joao/github/icub-main/bindings/python/_icub.so: undefined symbol: _ZN20ClientGazeControllerC1ERN4yarp2os10SearchableE
```

Any ideas?

Tobias-Fischer, 2015-10-27:
Hi,
I've quickly tested doing the same, and I run into the same error. I guess this is because the generated `_icub.so` links against some other libraries which are not found in return. Can you try putting `/path/to/install/yarp/lib` and `/path/to/install/icub-main/lib` in your `LD_LIBRARY_PATH`? This should fix the problem I guess.

Furthermore, delete the following lines from `icub.i`, as they are obsolete (see https://github.com/robotology/QA/issues/33):
```
YARP_DECLARE_DEVICES(icubmod)
YARP_REGISTER_DEVICES(icubmod)
```

Best,
Tobias

CC @pattacini @drdanz

pattacini, 2015-10-30:
@jpiabrantes, is @Tobias-Fischer's suggestion a good workaround?

lornat75, 2015-10-30:
you can also try compiling with rpath; this has recently become default and fixes path problems in .so dependencies without the need to use LD_LIBRARY_PATH. in case you need to pull, recompile, after cleaning cmake cache.

jpiabrantes, 2015-10-31:
@lornat75 I have updated and recompiled `icub`. I still have the same error. Should I also update and recompile yarp?

Tobias-Fischer, 2015-10-31:
Yes, and install yarp with rpath as well.

jpiabrantes, 2015-11-02:
I tried to recompile yarp and I got an error that I already described on another github [issue](https://github.com/robotology/QA/issues/63). I will try this again when I fix this issue.

Tobias-Fischer, 2015-11-16:
@jpiabrantes: Any news on this?

jpiabrantes, 2015-11-16:
@Tobias-Fischer All good now thanks :+1:

Tobias-Fischer, 2015-11-16:
How did you solve it? Did you have to set `LD_LIBRARY_PATH`?

Tobias-Fischer, 2015-11-16:
I just tried this (again), and I am still running in the same issue as mentioned in the first post. Therefore, I reopen this ticket.

pattacini, 2016-02-12:
Stale issue ...
It'd be worth closing this and - if it's the case - add/edit/report more on proper tracking system.

closed 2020-12-15 by pattacini