robotology/community #67: How to use CartesianSolver with iCub legs?

hu-yue, 2015-10-19:
Hi,
I don't have any experience with the CartesianSolver and I would like to ask if it's possible to use the CartesianSolver for our icub legs and how.
As our iCubHeidelberg01 has only legs, is it possible to use the solver as in the[ arm example](http://wiki.icub.org/brain/icub-tutorials_2src_2iKin_2onlineSolver_2main_8cpp_source.html)  but with [iCubLegCartesianSolver](http://wiki.icub.org/brain/classiCub_1_1iKin_1_1iCubLegCartesianSolver.html#afef0e40970aeeb3734b3e9777aa8d788) , or should we customize the CartesianInterface as in the [tutorial](http://wiki.icub.org/brain/icub_anyrobot_cartesian_interface.html)?

What we would like to have is the two feet as end effectors and perform online inverse kinematics.

Thanks!

hu-yue, 2015-10-19:
cc @kevinastein

pattacini, 2015-10-19:
Hi @hu-yue 

Assuming your legs kinematics is "**v2.5**", you have first to make sure that the following conditions are met:

- `robotInterface` is running with the legs Cartesian controllers launched.
To have an idea look at [`iCubGenova01`](https://github.com/robotology/icub-main/tree/master/app/robots/iCubGenova01/cartesian) app directory where we set up configuration files for the arm controllers. The `icub_all.xml` should then contain pointers to those files. You'll find attached an example for such [`left_leg_cartesian.xml`](https://files.zenhub.io/5624ebba35d756d07673a40b).

- Do `yarp-config context --import cartesianSolver cartesianSolver.ini` to copy in your local path the configuration file and then edit the version to `v2.5`. The final file should look like as this [`cartesianSolver.ini`](https://files.zenhub.io/5624ec3535d756d07673a425).
Now you can launch the solver: `iKinCartesianSolver --part left_leg`.

Finally, within your code, you can do:
```cpp
Property option;
option.put("device","cartesiancontrollerclient");
option.put("remote","/icub/cartesianController/left_arm");
option.put("local","/client/left_arm");
PolyDriver clientCartCtrl(option);
ICartesianControl *icart=NULL;
if (clientCartCtrl.isValid()) {
   clientCartCtrl.view(icart);
}
```

hu-yue, 2015-10-19:
I've added the configuration files as suggested and it seems to work fine.
A question: when I perform a getPose, where exactly is the end effector? I've seen that it is possible to define different end effectors, but which frame is for example in the case of left_leg by default?
Thanks!

traversaro, 2015-10-19:
You can find the documentation of the DH chains used for the two legs in http://wiki.icub.org/wiki/ICubForwardKinematics , in particular http://wiki.icub.org/wiki/ICubWaistRightLegKinematicsV2.5 and http://wiki.icub.org/wiki/ICubWaistLeftLegKinematicsV2.5 . 
They are the frames that in the `iCubHeidelberg01` urdf are named as `l_foot_dh_frame` and `r_foot_dh_frame` (see http://wiki.icub.org/wiki/ICub_Model_naming_conventions ).

hu-yue, 2015-10-19:
I see, thanks! =)

pattacini, 2015-10-19:
@hu-yue there was a typo (fixed now) in my previous comment.

The correct line is:
```cpp
option.put("remote","/icub/cartesianController/left_leg");
```
I believe you figured it out anyway.

hu-yue, 2015-10-19:
Yes, no problem with the code! I'm always aware that we don't have arms! =)

pattacini, 2015-10-19:
Eheh :smile:

closed 2020-12-08 by pattacini