robotology/community #68: icubmod and icubmoddev have been

pattacini, 2015-10-22:


pattacini, 2015-10-22:
_icubers_ take note of this latest change.

pattacini, 2015-10-22:
As @Tobias-Fischer correctly pointed out, it is also required to remove the following lines from the C++ code:

```cpp
YARP_DECLARE_DEVICES(icubmod)
YARP_REGISTER_DEVICES(icubmod)
```

traversaro, 2015-10-23:
@pattacini @mbrunettini can we increase the minor version number of icub-main such that we can cmake depend on the latest version of `icub-main` (after this modification)?

traversaro, 2015-10-23:
Even the patch version was ok, thanks!

pattacini, 2015-10-23:
Semantically speaking, it was not a patch but a sort of change in the "_interface_", hence "minor" is more appropriate.

traversaro, 2015-10-23:
I was just scared since the minor version was not updated since 2010 : https://github.com/robotology/icub-main/tags .

pattacini, 2015-10-23:
Time has come to update it, then :smirk: 
I'm fine with changing the patch too, so whoever admin wants to kick in and do that, there's no problem.

jeljaik, 2015-10-23:
Bit more details can also be found here: https://github.com/robotology/QA/issues/33#issue-81497604
The original post is missing **`YARP_DECLARE_PLUGINS(icubmod)`** and **`YARP_REGISTER_PLUGINS(icubmod)`**.

pattacini, 2015-10-24:
Thanks @jeljaik; I've updated the post.

closed 2020-12-12 by pattacini