robotology/community #72: Compiling the icub_firmware_shared

iron76, 2015-10-28:
I am trying to compile the icub-firmware-shared so as to be able to finally compile the `canLoader` to access some FT sensors with the CAN-USB device from ESD (see also #60). Compilation of the icus-firmware-shared project is somehow obscure to me since nothing happens. Here is the cmake output:
```
-- 
-- The following features have been enabled:

 * canProtocolLib , Can Protocol Library.
 * embobj , EmbObj Library.

-- Configuring done
-- Generating done
-- Build files have been written to: /Users/iron/Code/icub-firmware-shared/build
```

and the make output:

```
/usr/local/Cellar/cmake/3.3.2/bin/cmake -H/Users/iron/Code/icub-firmware-shared -B/Users/iron/Code/icub-firmware-shared/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/local/Cellar/cmake/3.3.2/bin/cmake -E cmake_progress_start /Users/iron/Code/icub-firmware-shared/build/CMakeFiles /Users/iron/Code/icub-firmware-shared/build/CMakeFiles/progress.marks
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f CMakeFiles/Makefile2 all
make[1]: Nothing to be done for `all'.
/usr/local/Cellar/cmake/3.3.2/bin/cmake -E cmake_progress_start /Users/iron/Code/icub-firmware-shared/build/CMakeFiles 0
```

traversaro, 2015-10-28:
icub-firmware-shared is an header only library, just install it or use it from the build directory defining the icub_firmare_shared_DIR enviromental variable.

drdanz, 2015-10-28:
@traversaro Installation was removed by commit robotology/icub-firmware-shared@0da92ea6ea3c922e82b51e7aa29acf514235ca15 (I'm still planning to revert that commit as soon as @randaz81 is not looking :grin:). At the moment the only option is to use the build directory.

closed 2020-12-08 by pattacini