robotology/community #73: Information about company that produces iCub's flexible pcb cards

plinioMoreno, 2015-10-29:
Here in Lisbon we want to request a flexible pcb like the one the iCub has on its
fingertips and skin sensor (cdc004 in the iCubPlatform repository), in
order to put some hall sensors on Vizzy's hands.

We want to know where did you order the boards. And the price range,
because we want to pay a reasonable price.

pattacini, 2015-10-29:
Perhaps @maggia80 has info on that.

maggia80, 2015-10-29:
You can order it from PHOENIX S.R.L.  info at phoenixpcb.it. The price depends quite a lot from the size, the number of layers and the technology used (trace dimensions, hole sizes, etc.). Therefore you should send the gerber to them to get a quotation.

plinioMoreno, 2015-10-29:
Thanks @maggia80 for the info.

closed 2020-12-15 by pattacini