robotology/community #75: Enabling coloured prints in terminals

gsaponaro, 2015-11-05:
Hi,
what flags should I enable in Linux (i.e., environment variables, bashrc or yarp CMake flags) in order to make the yarp [INFO], [WARNING] etc. prints coloured, like in the screenshot posted at https://github.com/robotology/icub-support/issues/102 ?

Tobias-Fischer, 2015-11-05:
Adding `force_color_prompt=yes` in .bashrc should work.

gsaponaro, 2015-11-05:
Hi @Tobias-Fischer , thanks.

Well, we have that flag on our pc104 already. In the terminal I see colours when I type commands like ```ls```, but not on those yInfo() etc. prints.

Here is our .bash_iCubrc:
https://gist.github.com/gsaponaro/afaf4752b1b2f28334e5

I noticed that in that file we already have the following flags including ```COLORED_OUTPUT=1```, but still see no colours:
```
# DebugStream customization
export ICUB_VERBOSE_OUTPUT=0
export ICUB_COLORED_OUTPUT=1
export ICUB_TRACE_ENABLE=0
```

Furthermore, here is our yarp CMake cache:
https://gist.github.com/gsaponaro/142bf660094d698959c6

/cc @drdanz @mbrunettini

barbalberto, 2015-11-05:
yInfo macro is in yarp now, so I guess the correct variables to use are 
export YARP_VERBOSE_OUTPUT=0
export YARP_COLORED_OUTPUT=1
export YARP_TRACE_ENABLE=0

gsaponaro, 2015-11-05:
Thanks @barbalberto, that was it!

Is there documentation for DebugStream available?

I found two pages on the wiki mentioning this feature:
1. http://wiki.icub.org/wiki/Generic_iCub_machine_installation_instructions which is up-to-date including the correct variables
2. http://wiki.icub.org/wiki/ICub_PC104_Linux_Image_-_Detailed_info which still mentions the obsolete variables -> you may want to update that

mbrunettini, 2015-11-05:
My fault.
I forgot to update the 2.
Matteo

barbalberto, 2015-11-05:
Just did it, thank you for reporting.

barbalberto, 2015-11-05:
I did it, thank you for reporting.

gsaponaro, 2015-11-05:
Great, thanks.

One more related question: what does ```YARP_FORWARD_LOG_ENABLE``` do? I.e., to where are those logs forwarded?

If you want I can open another issue in the yarp repository to request this kind of documentation (if this does not exist already). Would appreciate it, it's not urgent for us but these seem like nice features to speed up debugging.

pattacini, 2015-11-10:
@gsaponaro I think the forwarding has to do with the **`yarplogger`** utility.

@randaz81 @barbalberto correct me if I'm wrong.

jeljaik, 2015-11-11:
@randaz81 might want to document this somewhere, but he had replied to this question [here](https://github.com/robotology/yarp/issues/392) by saying:

> Even if launching a module with yarprun is generally preferred in our lab, you can use the logger also if you start the module from the command line. In this case, you just need to set the environment variable YARP_FORWARD_LOG_ENABLE=TRUE on your machine

pattacini, 2015-11-11:
@jeljaik great cross-link! :+1: 

@randaz81 I know you're building a light saber at the moment, but once done it would be worth documenting `yarplogger` (way more prosy task I know ...) :smirk: 

... or assign this task to someone else (not to me!)

gsaponaro, 2015-11-12:
I opened another issue for that, closing this one, thanks.

closed 2020-12-08 by pattacini